google/gousb #3: Feature request - Support detach_kernel_driver function

nicovell3, 2017-06-24:
I need a handler on the gousb library to call the linux function "libusb_detach_kernel_driver", just as jpoirier does in his library (https://github.com/jpoirier/gousb/blob/master/usb/kerneldriver_linux.go). This will prevent my program from giving me a `libusb: device or resource busy [code -6]` error when I call `dev.Config(1)`. For now, I have to detach the device driver by hand to be able to run my program.

Should It try to perform a pull request with the addition of that code to this library? Or could someone implement it please?

Thanks in advance.

zagrodzki, 2017-06-24:
See https://godoc.org/github.com/google/gousb#Device.SetAutoDetach

nicovell3, 2017-06-24:
Hi zagrodzki. I have already tried that function and I was calling it. I'm attaching part of my code:

	// Pick the first device found.
	dev := devs[0]
	err = dev.SetAutoDetach(true)
	if err != nil {
	    log.Fatalf("%s.SetAutoDetach(true): %v", dev, err)
	}

	//dev.DetachKernelDriver()
	//defer dev.AttachKernelDriver()

	//Need to call libusb_detach_kernel_driver first
	// Switch the configuration to #1.
	cfg, err := dev.Config(1)

And when I run the `dev.Config(1)` line, it breaks with the error I mentioned, `libusb: device or resource busy [code -6]`.

By now, I've already ported the kernel_driver code from jpoirier to this library and when I uncomment the `dev.DetachKernelDriver()`, my application works perfectly. Could this code be added to the repository, please? In that case, I could make the pull request...

zagrodzki, 2017-06-24:
I see. I checked libusb and it seems like detach is called only when an interface is claimed, which doesn't occur when you switch the config. I would like to think this is a bug in libusb that we could fix. Let me see what I can find there.

Although, TBH, looking at the libusb code, I could find more issues with the detachment code there.

In the meantime - I don't think we really need to introduce a new interface. How about we call libusb detach on Config() if autodetach is set? That would actually make auto detach behavior correct for gousb, without modifying the user-visible surface.

Looking at the libusb code, handle->auto_detach_kernel_driver is used only in op_claim_interface and op_release_interface. That suggests we could skip the libusb_set_auto_detach_kernel_driver entirely in gousb, and instead just call attach/detach manually on config claim if autodetach is set.
As an added bonus, that would actually allow us whether reattach is needed or not, based on the result of detach call. And it would avoid the weird interaction where releasing an interface results in reattaching the kernel driver when you switch configurations.

Does that sound ok for your use case? Can you structure your changes around the internals of gousb's SetAutoDetach?

zagrodzki, 2017-06-25:
My reading of http://comments.gmane.org/gmane.comp.lib.libusb.devel.general/21876 is that the dev team is aware of the problem, but there are no plans to change the behavior. Manually detaching the driver on config change seems like the best option, I think.

nicovell3, 2017-06-25:
This is my first time developing with the libusb interface, but I'll try to do my best. I'll make a pull request soon structuring my changes around the internals of gousb.

zagrodzki, 2017-07-21:
Resolved by PR #6

closed 2017-07-21 by zagrodzki