google/gousb #13: Provide means of getting device's serial number

rojer, 2017-09-01:
In my practice, it is often necessary to distinguish between several devices with the same VID, PID.
The standard way of doing that is by serial number, however i don't see any way of getting it.
`Context.OpenDeviceWithVIDPIDSerial` would be ideal, but if `DeviceDesc` provided the `SerialNumber` field, then filtering in the `OpenDevices` callback would also be acceptable.
Even post-filtering of opened devices would be ok.
In any case, there needs to be a way to query device's serial number, that's the main thing.

rojer, 2017-09-02:
#14 is my minimally invasive stab at solving it. this allows me to proceed, whether it's the right way or not i have no idea :)

zagrodzki, 2017-09-02:
We've recently had an attempt at adding similar functionality in #9, but the author abandoned the change. Perhaps you'd like to pick up from where he left off?

I'd prefer to have one large extension to the library for covering related functionality, rather than small one-off changes.

Ideally I'd see a set of changes for retrieving all information from string descriptors, including device serial/manufacturer/product description, as well as configuration and interface descriptions, plus simple unit tests. See the comments in pull req #9 for more details.

rojer, 2017-09-02:
@zagrodzki oh, i see. looks like we ended up doing essentially the same thing, but my predecessor gave up on the unit test part.

i've adopted the interface of the final commit and added unit tests (thanks for the guidance you provided, it wasn't much hassle at all; i suspect that just like me @bernhardgoedel was using an older version of go so tests wouldn't build initially, i had to use more recent version of Go).
as far as config and interface... i'm sorry, but just like my predecessor i just don't have much time to invest into this, and i have no need to obtain string descriptions of configurations or interfaces.
however, my understanding is that this change would not conflict with that future addition, so my preference would be to have this committed and the rest left for later.

zagrodzki, 2017-09-03:
looking at your pull req, the only thing left to do for config/interface descriptions is to add index feld and an appropriate wrapper around GetStringDescriptor, I think that's fine then.

For the record, not having much time is a weak argument. Any additions of new features to the library create a maintenance burden on our side. Sometimes it's better in the long run not to accept a new feature at all than to accept a half done or poorly done implementation. But in this case I think the part you've done is a sufficient base for the complete feature set. The remaining part is just the API change, and I've had doubts about the best API to express the missing part, so it's probably fine, I can tackle it separately.

rojer, 2017-09-03:
thanks for your understanding. the question is always one of balance. we all have limited brain capacity and time (i am under very tight schedule atm). i judged that only implementing these 3 methods that i understand would be a reasonable step which will achieve a useful result and also save me time figuring out what the hell configs and interfaces are - something which i have no need nor interest in.
anyway, it looks like we found a workable compromise, which is good.

the library otherwise suits me really well, i'm glad Google stepped in to develop the (hopefully) libusb wrapper to rule them all :)

rojer, 2017-09-03:
btw, former Googler here :)

closed 2017-09-03 by zagrodzki