google/gousb #20: Device exists but returns "not_found" error

rlj1202, 2017-09-24:
```go
devices, err := ctx.OpenDevices(func (desc *gousb.DeviceDesc) bool {
		if desc.Vendor == gousb.ID(0x04e8) && desc.Product == gousb.ID(0x6860) {
			fmt.Println("found!")
			return true
		}
		return false
	})
```

This code prints a "found!" message. But `len(devices)` is zero and "not_found" error occurred. Is there a limit of supporting devices? Or Is there a bug? The device I want to communicate with is Samsung Galaxy J7 2016.

    panic: libusb: not found [code -5]

zagrodzki, 2017-09-24:
Looking through the gousb code that is used in this particular path, I believe there's only one case where that happens, and this is if libusb C call "libusb_open()" returns a "not found" error. Which is unusual, because that error is not on the list of the possible errors returned by libusb_open:
http://libusb.sourceforge.net/api-1.0/group__dev.html#ga8163100afdf933fabed0db7fa81c89d1

Looking in turn at the libusb code, that suggests the "not found" was returned from the OS-specific implementation. What platform are you running this on?

Please check the debug output that you get after setting debug level (https://godoc.org/github.com/google/gousb#Context.Debug) to 3 (info) or 4 (debug).

rlj1202, 2017-09-24:
I use Windows 10 64 bit and msys2 to use mingw, libusb 1.0 and pkg-config.

zagrodzki, 2017-09-24:
the debug output would be helpful, without it I'm completely in the dark.
I can only suggest that you check that you've installed the necessary drivers, as explained here: https://github.com/libusb/libusb/wiki/Windows#Driver_Installation
My experience with Windows/libusb is very limited though, so if that doesn't help, I'm not sure I'll be able to help at all.

For confirmation that it's not a problem with gousb, you might want to write a short C example that would open the device: try using https://github.com/libusb/libusb/blob/master/examples/listdevs.c as the base, and around line 38 add a call to libusb_open() and check the return value. If I'm correct, then it will also return -5 and the problem lies entirely within libusb.

rlj1202, 2017-09-25:
2232:1008 (bus 2, device 3): -12 path: 1.4
    04e8:6860 (bus 1, device 4): -12 path: 1.2
    046d:c52b (bus 1, device 3): -12 path: 1.1
    8087:0024 (bus 2, device 2): -12 path: 1
    8087:0024 (bus 1, device 2): -12 path: 1
    04e8:6860 (bus 1, device 4): -12 path: 1.2
    046d:c52b (bus 1, device 3): 0 path: 1.1
    0781:5571 (bus 1, device 6): -12 path: 1.5
    8086:1c26 (bus 1, device 1): -12
    2232:1008 (bus 2, device 3): -5 path: 1.4
    04e8:6860 (bus 1, device 4): -12 path: 1.2
    04e8:6860 (bus 1, device 4): -12 path: 1.2
    8086:1c2d (bus 2, device 1): -12
    8086:0189 (bus 1, device 5): -12 path: 1.3

Well, the only one which is opened successfully is mouse(HID) and the others are unsupported or not found. The device I want to communicate with is `04e8:6860` and this device returns error code `-12`. But in gousb, it returns `-5`.

Here is a full console debug output.

```go
ctx := gousb.NewContext()
defer ctx.Close()

ctx.Debug(4)

devices, err := ctx.OpenDevices(func (desc *gousb.DeviceDesc) bool {
	return desc.Vendor == gousb.ID(0x04e8) && desc.Product == gousb.ID(0x6860)
})
```

    [ 0.014009] [000029a0] libusb: debug [libusb_get_device_list] 
    [ 0.014009] [00003c9c] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
    [ 0.014009] [00003c9c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
    [ 0.014009] [00003c9c] libusb: debug [handle_events] poll fds modified, reallocating
    [ 0.014009] [00003c9c] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
    [ 0.015011] [000029a0] libusb: debug [windows_get_device_list] allocating new device for session [24B]
    [ 0.015011] [000029a0] libusb: debug [windows_get_device_list] allocating new device for session [90]
    [ 0.018012] [000029a0] libusb: debug [get_api_type] driver(s): usbhub
    [ 0.018012] [000029a0] libusb: debug [get_api_type] matched driver name against HUB API
    [ 0.018012] [000029a0] libusb: debug [windows_get_device_list] allocating new device for session [312]
    [ 0.019013] [000029a0] libusb: debug [get_api_type] driver(s): usbhub
    [ 0.020013] [000029a0] libusb: debug [get_api_type] matched driver name against HUB API
    [ 0.020013] [000029a0] libusb: debug [windows_get_device_list] allocating new device for session [4]
    [ 0.021014] [000029a0] libusb: debug [get_api_type] driver(s): usbhub
    [ 0.021014] [000029a0] libusb: debug [get_api_type] matched driver name against HUB API
    [ 0.021014] [000029a0] libusb: debug [windows_get_device_list] allocating new device for session [6]
    [ 0.022015] [000029a0] libusb: debug [get_api_type] driver(s): usbhub
    [ 0.022015] [000029a0] libusb: debug [get_api_type] matched driver name against HUB API
    [ 0.022015] [000029a0] libusb: debug [windows_get_device_list] allocating new device for session [210]
    [ 0.024017] [000029a0] libusb: debug [windows_get_device_list] allocating new device for session [37C]
    [ 0.024017] [000029a0] libusb: debug [init_device] got bus number from ancestor #4
    [ 0.114408] [00003c9c] libusb: debug [handle_events] poll() returned 0
    [ 0.114408] [00003c9c] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
    [ 0.114408] [00003c9c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
    [ 0.114408] [00003c9c] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
    [ 0.120937] [000029a0] libusb: debug [init_device] found 1 configurations (active conf: 1)
    [ 0.120937] [000029a0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 1261 bytes)
    [ 0.120937] [000029a0] libusb: debug [init_device] (bus: 2, addr: 3, depth: 1, port: 4): '\\.\USB#VID_2232&PID_1008&MI_00#7&35941262&0&0000'
    [ 0.122162] [000029a0] libusb: debug [windows_get_device_list] allocating new device for session [350]
    [ 0.122162] [000029a0] libusb: debug [init_device] got bus number from ancestor #4
    [ 0.122162] [000029a0] libusb: debug [init_device] found 2 configurations (active conf: 2)
    [ 0.123163] [000029a0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 47 bytes)
    [ 0.123163] [000029a0] libusb: debug [cache_config_descriptors] cached config descriptor 1 (bConfigurationValue=2, 113 bytes)
    [ 0.123163] [000029a0] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 2): '\\.\USB#VID_04E8&PID_6860&MS_COMP_MTP&SAMSUNG_ANDROID#7&12639422&0&0000'
    [ 0.124166] [000029a0] libusb: debug [windows_get_device_list] allocating new device for session [62]
    [ 0.124166] [000029a0] libusb: debug [init_device] found 1 configurations (active conf: 1)
    [ 0.126165] [000029a0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 84 bytes)
    [ 0.126165] [000029a0] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 1): '\\.\USB#VID_046D&PID_C52B&MI_00#7&13A163DB&0&0000'
    [ 0.127168] [000029a0] libusb: debug [windows_get_device_list] found existing device for session [4] (2.0)
    [ 0.127168] [000029a0] libusb: debug [init_device] got bus number from ancestor #2
[... 117 lines cut ...]
    [ 0.171199] [000029a0] libusb: debug [unsupported_open] unsupported API call for 'open' (unrecognized device driver)
    [ 0.171199] [000029a0] libusb: debug [libusb_open] open 1.4 returns -12
    [ 0.171199] [000029a0] libusb: debug [libusb_unref_device] destroy device 1.4
[... cut ...]
    [ 0.171199] [000029a0] libusb: debug [libusb_unref_device] destroy device 1.5
    [ 0.171199] [000029a0] libusb: debug [libusb_unref_device] destroy device 1.2
    [ 0.171199] [000029a0] libusb: debug [libusb_unref_device] destroy device 1.1
    [ 0.171199] [000029a0] libusb: debug [libusb_unref_device] destroy device 1.0
    [ 0.215231] [00003c9c] libusb: debug [handle_events] poll() returned 0
    [ 0.215231] [00003c9c] libusb: debug [libusb_exit] 
    [ 0.215231] [00003c9c] libusb: debug [libusb_exit] destroying default context

zagrodzki, 2017-09-25:
These logs and the errors from C libusb definitely point to lack of the drivers. The only APIs reported in this log is the HUB API and the Composite API, while based on the code I'd expect to see an entry with "WinUSB-like APIs", which would match either libusbK, libusb0 or WinUSB drivers. I think the difference between -5 (not found) and -12 (not supported) in gousb may be safely ignored for now, just get the C libusb to open your device successfuly.

Please follow the instructions in https://github.com/libusb/libusb/wiki/Windows#how-to-use-libusb-on-windows to install the required drivers.

rlj1202, 2017-09-25:
Thank for your kind answer! I tried many of drivers but neither one was suitable perfectly for my phone. Maybe I have to use Windows SDK.

closed 2017-09-25 by rlj1202

zagrodzki, 2017-09-25:
Note: these drivers are not "phone drivers" in any case. They are drivers that allow libusb raw access to the USB devices (generic USB, not any particular type of the USB device). That means installing e.g. libsubK (http://libusbk.sourceforge.net/UsbK3/index.html).