google/gousb #21: libusb: bad access [code -3]

minhajakhterkhan199, 2017-10-01:
I am getting ```libusb: bad access [code -3]```, please find source code below. It's not returning any open device, even when I am returning true.  devices are always 0... Please advise to fix the issue
```
ctx := gousb.NewContext()
	defer ctx.Close()

	devices, err := ctx.OpenDevices(func (desc *gousb.DeviceDesc) bool {
		if strings.Contains(usbid.Describe(desc),"SM-USB-DIG (Texas Instruments, Inc.)") {
			return true
		}
		return false
	})

	// All Devices returned from OpenDevices must be closed.
	defer func() {
		for _, d := range devices {
			d.Close()
		}
	}()

	for _, dev := range devices {
		// Once the device has been selected from OpenDevices, it is opened
		// and can be interacted with.
		manufacturer, err := dev.Manufacturer()
		if err!=nil {
			fmt.Println("err: %s",err)
		}else {
			fmt.Println("Manufacturer: %s",manufacturer)
		}

		//fmt.Printf("%03d.%03d %s:%s %s\n", dev.Bus, desc.Address, desc.Vendor, desc.Product, usbid.Describe(desc))
		//fmt.Printf("  Protocol: %s\n", usbid.Classify(desc))

		_ = dev
	}

	// OpenDevices can occaionally fail, so be sure to check its return value.
	if err != nil {
		log.Fatalf("list: %s", err) 
	}```

zagrodzki, 2017-10-01:
Please send the output with ctx.Debug(4) added before a call to OpenDevices. What platform are you running on?

minhajakhterkhan199, 2017-10-01:
development environment is Ubuntu 64. Please find the output below

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.732632] [0000199a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.732665] [0000199a] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.732691] [0000199a] libusb: debug [handle_events] poll() 2 fds with timeout in 100ms
[ 0.733536] [000019a0] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 3.806486] [0000199a] libusb: debug [handle_events] poll() returned 0
[ 4.822469] [0000199a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 4.822494] [0000199a] libusb: debug [handle_events] poll() 2 fds with timeout in 100ms
[ 4.822474] [00001993] libusb: debug [libusb_get_device_list] 
[ 4.822594] [00001993] libusb: debug [libusb_get_device_descriptor] 
[ 4.822637] [00001993] libusb: debug [libusb_get_config_descriptor] index 0
[ 5.399340] [0000199a] libusb: debug [handle_events] poll() returned 0
[ 5.916213] [0000199a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 5.916229] [0000199a] libusb: debug [handle_events] poll() 2 fds with timeout in 100ms
[ 7.147556] [0000199a] libusb: debug [handle_events] poll() returned 0
[ 7.147578] [0000199a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 7.147582] [0000199a] libusb: debug [handle_events] poll() 2 fds with timeout in 100ms
[ 7.147745] [00001993] libusb: debug [libusb_open] open 1.5
[ 7.147822] [00001993] libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/005: Permission denied
[ 7.147827] [00001993] libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[ 7.147829] [00001993] libusb: debug [libusb_open] open 1.5 returns -3
[ 9.406398] [0000199a] libusb: debug [handle_events] poll() returned 0
[ 9.406423] [0000199a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 9.406426] [0000199a] libusb: debug [handle_events] poll() 2 fds with timeout in 100ms
[ 9.409305] [00001993] libusb: debug [libusb_get_device_descriptor] 
[ 9.409330] [00001993] libusb: debug [libusb_get_config_descriptor] index 0
[10.896267] [00001993] libusb: debug [libusb_get_device_descriptor] 
[10.896296] [00001993] libusb: debug [libusb_get_config_descriptor] index 0
[10.898191] [0000199a] libusb: debug [handle_events] poll() returned 0
[10.898284] [0000199a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[10.898293] [0000199a] libusb: debug [handle_events] poll() 2 fds with timeout in 100ms
[11.809498] [0000199a] libusb: debug [handle_events] poll() returned 0
[11.809520] [0000199a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[11.809525] [0000199a] libusb: debug [handle_events] poll() 2 fds with timeout in 100ms
[14.275903] [0000199a] libusb: debug [handle_events] poll() returned 0
[14.275981] [0000199a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[14.275985] [0000199a] libusb: debug [handle_events] poll() 2 fds with timeout in 100ms
[16.427548] [0000199a] libusb: debug [handle_events] poll() returned 0
[16.427600] [0000199a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[16.427606] [0000199a] libusb: debug [handle_events] poll() 2 fds with timeout in 100ms
[17.309986] [0000199a] libusb: debug [handle_events] poll() returned 0
[17.310024] [0000199a] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[17.310028] [0000199a] libusb: debug [handle_events] poll() 2 fds with timeout in 100ms
2017/10/01 13:08:02 list: libusb: bad access [code -3]

zagrodzki, 2017-10-01:
This is the relevant part:

```
[ 7.147822] [00001993] libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/005: Permission denied
[ 7.147827] [00001993] libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[ 7.147829] [00001993] libusb: debug [libusb_open] open 1.5 returns -3
```

The user you're running as does not have permissions to access that USB device. You'll need to update the udev rules or change device permissions in some other way.

minhajakhterkhan199, 2017-10-01:
Thanks for your prompt response issue fixed! 
Kindly give me example / source code, how to write & read data on USB interface.
In this regards, waiting for your response.

Best Regards,

zagrodzki, 2017-10-01:
See https://godoc.org/github.com/google/gousb#pkg-examples and https://github.com/google/gousb/tree/master/rawread for examples.

closed 2017-10-01 by zagrodzki

filinvadim, 2020-03-12:
BTW Mac has same problem but it hasn't udev rules