google/gousb #25: Feature Request: Async API

konkers, 2017-11-13:
Has there been thought put into an Async API?  I'd like to be able to queue an RX transaction and either receive notification or the packet itself on a channel.

zagrodzki, 2017-11-13:
I see, I think that's a valid concern.
I don't think we would implement an async API, but perhaps we should add variants to some methods ( ep.Read/Write and Device.Control come to mind) that would accept a context.Context to control cancellation.

In the meantime, would the streaming interface work better for you? ReadStream has a Close() method that will abort transfers in flight. I expect it would work if you're doing repeated reads, it will not work well for a one-off read.

We also used to have a Timeout field on the endpoint, but I see now that during refactoring we accidentally hid it from the public view (it's now a memeber of an unexported struct). I could restore Timeout as a field on InEndpoint, which would at least let you control how long the Read will wait for the transaction. Would that be of any help?

konkers, 2017-11-15:
I've been playing around with ReadStream.  I think it fits my needs pretty well at the moment.  It does have some problems surrounding timeouts.  Once one happens, the whole stream closes.  I have to do a sync dance with my device in order to synchronize the endpoint data toggles on open.  I'm using the non-streaming Read calls to do then, then open the stream.  I'm not far enough along to know if I need to handle timeouts after the first sync.

Methods that take context.Context would also work and have a nice parallel with the network handler stack.

zagrodzki, 2017-11-15:
@stapelberg @kylelemons any thoughts on adding context-enabled Read/Write/Control?
I'd rather not modify the existing API (and it would no longer be compatible with io.Reader etc).
So, perhaps ContextRead/ContextWrite on respective In/OutEndpoint?

I expect the implementation to be simple, since the transfer used for all of the above supports explicit cancellation.

zagrodzki, 2017-11-15:
an alternative would be something similar to https://godoc.org/github.com/jbenet/go-context/io, NewReader/Writer. That would allow use of context for long-lived operations rather than individual Read/Writes.

stapelberg, 2017-11-16:
Either of the two approaches sounds good to me.

rbredlau, 2018-02-17:
I've been using the deadsy/libusb API and just started investigating this one.  I like this library better because the types are easier to use and performing reads with a ReadStream doesn't wake up the CPU constantly.  However my devices are keypads and card readers, thus dependent on user action, and the blocking Read() prevents a clean program shutdown.  A cancellable API or even some form of timeout would be greatly appreciated.

In the meantime I've found a work-around that will cause Read() to return early when I want it to.  Simply call the done() method returned when obtaining the interface and calls to Read() return with an error that the device is not available.

zagrodzki, 2018-11-05:
PR #57 added the cancellable context support for reads and writes. For control, Device.ControlTimeout may be used. In the future we could switch Control to ControlContext as well, but this would require replacing the use of convenience libusb_control_transfer function with more explicit manual transfer construction, so perhaps we don't need it right away.

closed 2018-11-05 by zagrodzki