google/gousb #28: Feature Request:  Support for libusb_get_parent & libusb_get_port_number

djadala, 2017-12-01:
Hi,
I would like to ask for support for functions ```libusb_get_parent``` and ```libusb_get_port_number```.
I want to be able to print devices as tree.
To start, i guess i can submit PR for  ```libusb_get_port_number```  soon.
Regards

zagrodzki, 2017-12-01:
libusb_get_port_number results are already implemented (the device desc has a "Port" field).

Some notes for implementing the "tree": 

Using libusb_get_parent might be tricky because of the requirement to use it only between "get list" and "free list" (both of which now happen in libusb.getDevices). Otherwise I understand the pointer to the parent might be not the same pointer that you have on the list and things get confusing.

There is libusb_get_port_numbers that returns a "port number path" from the bus root, where the last number should be the same as the port number from libusb_get_port_number. That looks like something easy to implement and we could put the path in the device descriptor struct.

The "Open..." function iterates over all devices. Perhaps it might make sense to introduce an explicit ordering top-bottom for the iteration, so that when the opener function is called on each device, you get to inspect a parent before inspecting all of its children. By comparing bus numbers and checking if some path is a prefix of device path, you can establish the parent-child relationship without referring to libusb_get_parent.

djadala, 2017-12-01:
Thank  You for detailed explanation, i don't have experience working  with usb, so your notes are very helpful.
Can some info for parent device can be included in DeviceDesc struct, ? for example parent bus & address ?
Thank again.

zagrodzki, 2017-12-01:
Parent bus is the same as the device bus, by definition. Each bus has a "root" device (typically a root hub) that itself doesn't have a parent. All devices on the bus form a tree under this root. Each of the devices has exactly one path through the tree of ports, this is the structure returned by libusb_get_port_numbers. The path of the root is empty ([]), the path of a device connected to root's port 1 is [1], the path of a device connected to port 3 of a device connected to root's port 1 is [1,3] etc.

That means we have at least two way to reference a particular device in the device structure - by device address (which are just numbers), or by path (a sequence of port numbers). It's difficult for me to say which would be more useful. I suspect reference by address might be better for the API, while the getDevices implementation probably should use the port path to ensure the correct ordering (sort by bus + port path, essentially).

djadala, 2017-12-02:
Thank again,
as can i see, libusb_get_port_numbers is sufficient to reconstruct device tree,
it can be added simple (not tested):

```go
func (d *Device) PortNumbers() ([]byte, error) {
	if d.handle == nil {
		return 0, fmt.Errorf("PortNumbers() called on %s after Close", d)
	}
	return d.ctx.libusb.portNumbers(d.handle)
}

func (libusbImpl) portNumbers(d *libusbDevHandle) ([]byte, error) {
    data [7]byte
	n := C.libusb_control_transfer(
		(*C.libusb_device_handle)(d),
		(*C.uint8_t)(unsafe.Pointer(&data[0])),
		C.int(len(data)))
	if n < 0 {
		return nil, fromErrNo(n)
	}
	return data[:n], nil
}
```
do you want PR ?

zagrodzki, 2017-12-02:
I don't think I want to have it as a separate call and as I've mentioned above, I'd like to keep the API close to what the user might want rather than what libusb provides.

I think it makes sense to add the parent device address in a new field in DeviceDesc, called "Parent".
We might add a port path in a separate field, "Path", but that is secondary to the parent/child relation I think.

To find and set the parent id gousb needs to construct the tree and populate the field in all device descriptors accordingly.

zagrodzki, 2018-05-23:
Is this something you want to keep working on? If not, we'll probably shelve it, until there's more interest from the users.

djadala, 2018-05-23:
Currently i don't have time,  sorry.

zagrodzki, 2018-05-23:
Noted, thanks. I'll close for now, we can revisit in the future if needed.

closed 2018-05-23 by zagrodzki

willmcenaney, 2021-10-20:
Resurrecting this as I have code to contribute to add Path to DeviceDesc