google/gousb #35: Better Control() interface with data

Julio-Guerra, 2018-04-09:
Hi,

I found several problems in passing data to `Device.Control()` as a slice of bytes:

1. Control requests are generally used to pass low-level data types, but since Go doesn't support packed structures, they are implemented as packed C types. But CGo doesn't support packed C structures either. Which means that it is impossible to read the data from this slice in any **standard** Go way. 
2. This byte slice is passed to the libusb by reading the `SliceHeader` structure, which is Go's implementation-defined.
3. When writing portable standard Go, it requires making at least two copies of the data: slice -> C struct -> optionally Go struct when C struct is not packed. 

Here is an example of a C struct that cannot be correctly accessed from Go - this is the one with I started having problems with current interface:
```
struct hub_descriptor {
    uint8_t  DescLength;
    uint8_t  DescriptorType;
    uint8_t  NbrPorts;
    uint16_t HubCharacteristics;
    uint8_t  PwrOn2PwrGood;
    uint8_t  HubContrCurrent;
} __attribute__((packed));
```
Its `HubCharacteristics` field appears as `_` in Go. It is thus ignored by `encoding/binary.Read()` function... For the record, C getter functions need to be written to read fields from a packed structure.

So I modified gousb to get rid of the 3 problems I listed previously. There are two implementation options I see:
1. Changing data's type to `interface{}` and use `unsafe.Pointer(reflect.ValueOf(data).Pointer())`. The advantage of this option is that we are now able to pass both slices and pointers to structures and int types, i.e. the two main types used with `Control()`. Note that it also requires one more argument to provide the size of `data`, since we cannot now the size of such types using `interface{}`. 
2. Changing data's type to `uintptr`, and letting users getting it.

I am personally using option 1 for now. If you guys agree, I can push the modifications in a PR.

zagrodzki, 2018-04-09:
I'm not sure I understand all that you're saying, in particular I don't get the "HubCharacteristics appears as _ in Go" - this suggests you're talking about some specific way of wrapping that structure in Go? And I'm not sure I understand the "low-level data types", unless again you're talking about wrapping some specific C implementation in gousb interface?

When working with USB, I tend to look primarily at USB spec/device spec rather than C structures that some code already uses to represent them. So for the case you've described (reading the hub descriptor), I'd define a Go structure (matching 1:1 what you showed above), using encoding/binary.Read to decode the first 7 bytes of the structure, then parse remaining data as necessary. Does that not work?

Julio-Guerra, 2018-04-09:
> I'm not sure I understand all that you're saying, in particular I don't get the "HubCharacteristics appears as _ in Go" - this suggests you're talking about some specific way of wrapping that structure in Go?

If you compile this structure packed, this field is not named `HubCharacteristics` but `_`, when read from Go. Thus leading to compile errors saying this field is not defined. With packed C structures, you also need to write getter to read them in C, and call them from Go. It is documented here: https://github.com/golang/go/wiki/cgo#struct-alignment-issues

> And I'm not sure I understand the "low-level data types", unless again you're talking about wrapping some specific C implementation in gousb interface?

By low-level data type, I mean here any structure that cannot be represented as a Go structure. Still because of the packing limitation. 

What you suggest - reading the slice by contiguous struct chunks - requires users to know Go's alignment rules: they are (at least) specific to the architecture. And this is why `AlignOf()` is in package `unsafe`.

Here is an example of how this field is aligned and thus not packed - its offset is 4, not 3: https://play.golang.org/p/KSYS-3X5RwG Between parsing a `[]byte` knowing Go's internal rules and simply reading a packed C struct, I choose the second one ;)

zagrodzki, 2018-04-09:
What you're saying applies, I believe, only to C structs and trying to use memory representation of the structure directly as the wire format. I don't think that's a good approach in general with Go.

I would suggest to use either a marshal/unmarshal (or encode/decode) approach for a particular structure, or using encoding/binary for a more direct approach.

So for marshal, that would go like this:

type PowerSwitching int
const (
  PowerSwitchingGanged PowerSwitching = iota
  PowerSwitchingIndividual
  PowerSwitchingNone
)

type HubDescriptor struct {
  NbrPorts int
  PowerSwitching PowerSwitching
  Compound bool
  OverCurrentProtection ...
  ThinkTime ... (not sure if that should be expressed in frames or time.Duration)
  IndicatorsSupported bool
  ...
  Removable map[int]bool
}

func (d *HubDescriptor) Unmarshal(b []byte) error {
  if len(b) < 7 { return fmt.Errorf(...) }
  d.NbrPorts = int(b[2])
  switch b[3] & 0b11 {
  case 0:
    d.PowerSwitching = PowerSwitchingGanged
  case 1:
   d.PowerSwitching = PowerSwitchingIndividual
  ...
  }
  d.Compound = b[3] & 0b100 != 0
  ...
}

etc, you get the idea. This very explicitly defines a byte wire representation of a structure that is usable in Go. This structure additionally is better in terms of logical representation of the functionality - why represent hub characteristics as uint16 with a bunch of maps when as a user you're interested in individual features represented by this structure.

If you care more about getting a struct that maps exactly to USB spec descriptor, then I'd suggest using a Go struct mapping 1:1 to spec and using encoding/binary.Read/Write combined with a bytes.Buffer as a generic marshaler/unmarshaler. Note that encoding/binary.Read/Write have nothing to do with memory layout or memory representation, they just write out data representation as bytes, exactly like you'd expect. See this example: https://play.golang.org/p/kuC9CglbTc_

I assume for non-standard control requests, users would need to write their own decoders. But for standard descriptors, as mentioned in the USB spec, I wouldn't have anything against adding their useful representations in gousb, if you'd like to submit the implementation for the ones you need.

zagrodzki, 2018-04-13:
That should be fine, especially considering that Value.Pointer does exactly the same operation underneath:
https://github.com/golang/go/blob/1dbe4c50f2e722765240c8f2fc261eab83802f99/src/reflect/value.go#L1296

Julio-Guerra, 2018-05-23:
No, I won't have any time to do it. I'll try to give you the HUB descriptors I wrote in the future.

closed 2018-05-23 by Julio-Guerra