google/gousb #37: warning: libusb_set_debug is deprecated

aletheia7, 2018-05-18:
### I am getting the following error with go install -v:
```sh
<redacted>/vendor/github.com/google/gousb
cgo-gcc-prolog: In function ‘_cgo_8c50a4ca37d7_Cfunc_libusb_set_debug’:
cgo-gcc-prolog:505:2: warning: ‘libusb_set_debug’ is deprecated: Use libusb_set_option instead [-Wdeprecated-declarations]
In file included from vendor/github.com/google/gousb/libusb.go:31:0:
/usr/include/libusb-1.0/libusb.h:1300:18: note: declared here
 void LIBUSB_CALL libusb_set_debug(libusb_context *ctx, int level);
                  ^~~~~~~~~~~~~~~~
  ```
### go env
```
GOARCH="386"
GOBIN=""
GOCACHE="/home/erik/.cache/go-build"
GOEXE=""
GOHOSTARCH="386"
GOHOSTOS="linux"
GOOS="linux"
GOPATH="/home/erik/go"
GORACE=""
GOROOT="/usr/lib/go-1.10"
GOTMPDIR=""
GOTOOLDIR="/usr/lib/go-1.10/pkg/tool/linux_386"
GCCGO="gccgo"
GO386="387"
CC="gcc"
CXX="g++"
CGO_ENABLED="1"
CGO_CFLAGS="-g -O2"
CGO_CPPFLAGS=""
CGO_CXXFLAGS="-g -O2"
CGO_FFLAGS="-g -O2"
CGO_LDFLAGS="-s -w"
PKG_CONFIG="pkg-config"
GOGCCFLAGS="-fPIC -m32 -pthread -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build827923295=/tmp/go-build -gno-record-gcc-switches"
```

aletheia7, 2018-05-18:
Linux: Debian Testing/Buster
libusb package: libusb-1.0-0:i386       2:1.0.22-2
gousb:  5115a55

zagrodzki, 2018-05-19:
To clarify: does this break the build (i.e. go build aborts with an error), or is it just a matter of build displaying a warning message?

If it's the latter, then I'd rather not make any changes in gousb to accomodate the new options - we would do that when we make the next large revision of the API, at which point I can simply require libusb >= 1.0.22 and don't need to define conditions based on libusb version.

If it's the former, I've updated the branch "set_debug" (https://github.com/google/gousb/tree/set_debug), would you mind trying it out? Travis offers me "trusty" as the newest ubuntu environment, which has much older libusb, my personal systems also don't have 1.0.22 yet. I could install a new system just for this, but since you already have an environment where it fails...

aletheia7, 2018-05-19:
It does not break the build. However, the extra warning output messes (the source disappears, filed issue: https://github.com/fatih/vim-go/issues/1832) with the vim-go plugin and editing, and consequently, other go files.

I will try set_debug today. Thank you.

aletheia7, 2018-05-20:
Almost there. Now getting:
```
go build -v
# <redacted>/vendor/github.com/google/gousb
cgo-gcc-prolog: In function ‘_cgo_843581783fcb_Cfunc_libusb_set_debug’:
cgo-gcc-prolog:518:2: warning: ‘libusb_set_debug’ is deprecated: Use libusb_set_option instead [-Wdeprecated-declarations]                           
In file included from vendor/github.com/google/gousb/libusb.go:28:0:
/usr/include/libusb-1.0/libusb.h:1300:18: note: declared here
 void LIBUSB_CALL libusb_set_debug(libusb_context *ctx, int level);
                  ^~~~~~~~~~~~~~~~
```

zagrodzki, 2018-05-20:
That doesn't sound right: the code in that branch doesn't call C.libusb_set_debug anymore (as indicated by Cfunc_libusb_set_debug in your snippet), it should now be calling C.gousb_set_debug (see libusb.go line 223). Can you make sure that the vendored code you're using was checked out from the set_debug branch?

aletheia7, 2018-05-21:
The set_debug branch works great. Thank you.

aletheia7, 2018-06-14:
I performed a git pull and my master is c6e780971f197b0c890e80224f4137868f139668.

The above problem has returned. The set_debug branch has disappeared, however I do have the has where your fix is. Any ideas what I have done wrong? Thank you.
```sh
$ go build -v
# <redacted ...>/vendor/github.com/google/gousb
cgo-gcc-prolog: In function ‘_cgo_843581783fcb_Cfunc_libusb_set_debug’:
cgo-gcc-prolog:518:2: warning: ‘libusb_set_debug’ is deprecated: Use libusb_set_option instead [-Wdeprecated-declarations]
In file included from ../vendor/github.com/google/gousb/libusb.go:28:0:
/usr/include/libusb-1.0/libusb.h:1300:18: note: declared here
 void LIBUSB_CALL libusb_set_debug(libusb_context *ctx, int level);
                  ^~~~~~~~~~~~~~~~
$ dpkg-query -W libusb\*
libusb-1.0-0:i386       2:1.0.22-2
libusb-1.0-0-dev:i386   2:1.0.22-2
libusb-1.0-doc  2:1.0.22-2
$ git status
On branch set_debug
Your branch is up to date with 'origin/set_debug'.

nothing to commit, working tree clean
$ git log | head -20
commit 9b217d35e1709b0e73fb7f48c3010dc748e3b030 (HEAD -> set_debug, origin/set_debug)
Author: Sebastian Zagrodzki <sebek@zagrodzki.net>
Date:   Sat May 19 18:57:23 2018 +0200

    Support libusb_set_option in additon to libusb_set_debug.
    The latter is deprecated in libusb 1.0.22 and later.

commit d0366361a25268248b2e53405b91eaa0deadf761
Author: Sebastian Zagrodzki <sebek@zagrodzki.net>
Date:   Sun Nov 12 03:32:09 2017 +0100

    Add missing arguments to Interface() error on failed descriptor
    retrieval.

commit b893afb0188fc0c9f9c3ae60bee6beb1b87af1b2
Author: Sebastian Zagrodzki <sebek@zagrodzki.net>
Date:   Sun Nov 12 03:28:47 2017 +0100

    Add missing endpoint argument to Fatalf.

```

zagrodzki, 2018-06-15:
I'm guessing that the version in ../vendor/..../gousb is different from the one you pulled. Your message says: "In function ‘_cgo_843581783fcb_Cfunc_libusb_set_debug". Current master doesn't have Cfunc_libusb_set_debug anymore, it would be Cfunc_gousb_set_debug. This means that contrary to what you think, you're using an older revision of the code.

aletheia7, 2018-06-15:
The only gousb I have is in my vendor directory.

The current master does have libusb_set_debug.
```sh
$ grep -nHr set_debug *
libusb.go:34:void gousb_set_debug(libusb_context *ctx, int lvl);
libusb.go:223:  C.gousb_set_debug((*C.libusb_context)(c), C.int(lvl))
libusb.go:503:  C.libusb_set_debug((*C.libusb_context)(c), C.int(lvl))
usb.c:18:void gousb_set_debug(libusb_context *ctx, int lvl) {
usb.c:24:    libusb_set_debug(ctx, lvl);
```
I was able to suppress the warning with the following code change:
```golang
func libusbSetDebug(c *libusbContext, lvl int) {                                                                                                                                                                     
    //#pragma GCC diagnostic push                                                                                                                                                                                    
    //#pragma GCC diagnostic ignored "-Wdeprecated-declaration"                                                                                                                                                      
    C.libusb_set_debug((*C.libusb_context)(c), C.int(lvl))                                                                                                                                                           
    //#pragma GCC diagnostic pop                                                                                                                                                                                     
}
```
I would prefer to use gousb master instead of my own branch.

zagrodzki, 2018-06-16:
I see, you're right. There are two places where libusb_set_debug used to be called, and I've updated only one of them. The other one is used only in tests, but it's part of the library, because using C in tests doesn't work well for some reason or other (as I recall). Updated the other one. Please check now. Sorry for confusion.

closed 2018-06-16 by zagrodzki