google/gousb #45: Doesn't work on ARM (rasbperry pi)

willemcvu, 2018-08-07:
I'm trying to build a package that includes gousb on a Raspberry Pi 2, and I'm unable to get gousb working at all.
`go get -v github.com/google/gousb` returns:
`github.com/google/gousb (download)
github.com/google/gousb
# github.com/google/gousb
../github.com/google/gousb/libusb.go:234:29: could not determine kind of name for C.libusb_get_port_number
../github.com/google/gousb/libusb.go:396:19: could not determine kind of name for C.libusb_set_auto_detach_kernel_driver
`
Any ideas?

zagrodzki, 2018-08-07:
Which version of Go and which version of libusb?

willemcvu, 2018-08-07:
`libusb-1.0-0 `and `go1.10.3`.
I also tried `go1.8`, same exact issue.

zagrodzki, 2018-08-07:
libusb-1.0-0 is not helpful, unfortunately, that's the name of the package (Debian or Ubuntu, I'm guessing), but not the version. dpkg -l libusb-1.0-0 should help.

willemcvu, 2018-08-07:
Whoops, sorry!
2:1.0.11-1 is the version I'm using (as returned from dpkg).

zagrodzki, 2018-08-07:
1.0.11 was released in May 2012, so I think it's fair to call it really old... Port number function was added few days later in May 2012, and kernel autodetach was added in June 2013. The first version that includes both was 1.0.16, released Jul 2013. Current version is 1.0.22.

If that is indeed Debian wheezy, you'll either need to install 1.0.19 from wheezy-backports or upgrade to jessie.

closed 2018-08-07 by zagrodzki

willemcvu, 2018-08-07:
OK, thanks a ton, I will try upgrading my distro.

zagrodzki, 2021-11-02:
@Gustavomurta I don't think your instructions help - first, the problem mentioned in this question was exactly that the `libusb` that came with the distribution (as installed by `apt-get` in your documentation) was too old and didn't have the required functions.

Second, your instructions are specific to Debian-flavor of Raspberry Pi (Raspbian etc) and not generally useful for other distributions (SUSE, Gentoo, Arch etc.).