google/gousb #47: Feature Request

bsakweson, 2018-09-04:
UBS plugged and unplugged events:

I am unable to locate plugged and unplugged events in this package. It would be nice to know if a usb device is plugged or unplugged so that implementors can better manage reconnection when an unplugged device is later plugged-in.

zagrodzki, 2018-09-04:
The package doesn't support hotplug at all at this point. There is some work done towards supporting hotplug in https://github.com/google/gousb/pull/32, but it's unfinished and didn't see any updates for a long time now. It's something that could be implemented if you'd like to work on it.

Elemecca, 2018-10-20:
This issue is a duplicate of #8 and should be closed as such.

closed 2018-11-05 by zagrodzki