google/gousb #65: ls usb example fails

aluedeke, 2019-04-15:
I get the following exeption on my Lenovo X1 when running the lsusb example

`2019/04/15 09:29:36 list: config 1 interface at index 0 has number 12, USB standard states they should be identical
exit status 1`

Running the ubuntu lsusb command works just fine. Could it be gousb is a bit over agressive here?

zagrodzki, 2019-05-06:
On one hand, you might be right, we probably shouldn't have added that assumption.

On the other hand, the assumption about sequential interface numbers is hardcoded at this point into the gousb interface:
https://godoc.org/github.com/google/gousb#ConfigDesc

You'll notice that ConfigDesc.Interfaces is a slice, not a map, and we expect the interface number to correspond to the index of the slice, i.e. descriptor of interface 0 is accessible through ConfigDesc.Interfaces[0].

I suspect we could change that - such change would not be backward incompatible (since this use case throws an error anyway) and we'd only verify that interface+alt setting combinations are unique. It would require some extra processing, but wouldn't need changes to the interface, I think.

Would you mind showing the output of `lsusb -v` where gousb fails?

IvoBCD, 2020-04-20:
@zagrodzki Also observed with NXP's LPC Link 2 debugger, with the following lsusb -v output:

```
Bus 001 Device 009: ID 1fc9:0090 NXP Semiconductors 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1fc9 NXP Semiconductors
  idProduct          0x0090 
  bcdDevice            1.00
  iManufacturer           1 NXP Semiconductors
  iProduct                2 LPC-LINK2 CMSIS-DAP V5.224
  iSerial                 3 KXBUMSIU
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          171
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              4 LPC-LINK2 CMSIS-DAP V5.224
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      35
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               4
    Interface Descriptor:
[... 137 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               2
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               2
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```

aluedeke, 2020-04-25:
my lsusb -v output looks like this:
https://gist.github.com/aluedeke/571b81aeb8cabf91302d1dc9ce56b01c

zagrodzki, 2020-04-26:
can you see if the fix in branch "noncompliant-interface-desc" works for you?

aluedeke, 2020-04-27:
@zagrodzki works like a charm

closed 2020-04-28 by zagrodzki