google/gousb #68: Control endpoint doesn't support packets

jennysharps, 2019-06-14:
The Control() method implemented on device https://github.com/google/gousb/blob/master/device.go#L195 doesn't seem to accept a "wLength" argument (see https://www.beyondlogic.org/usbnutshell/usb6.shtml).  As such, the device only receives the first packet being sent (8 bytes in my case as it's max allowed packet size).  It looks like the underlying C method being called already accepts an additional argument after the data arg, presumably for this purpose.

Endpoints exposed via an Interface also support similar behavior--from godocs: "An InEndpoint implements the io.Reader interface, an OutEndpoint implements the io.Writer interface. Both Reads and Writes will accept larger slices of data than the endpoint's maximum packet size, the transfer will be split into smaller USB transactions as needed".

Is there any reason passing in wLength hasn't been implemented in this lib?  Or is there an existing mechanism for transferring multiple packets to a control endpoint?

zagrodzki, 2019-06-17:
I don't think that is correct - wLength is used, it's set to the length of the data slice that is passed to the call. See https://github.com/google/gousb/blob/2dc560e6bea3d29e5bb5cc20c71dac201dc7541f/libusb.go#L349

The underlying C method, `libusb_control_transfer`, needs the `wLength` argument simply because in C the request data is passed as a pointer, with no inherent length; the library needs to be explicitly informed of the size of the transfer buffer.

If I remember correctly, all of the methods implemented in this library rely on the host handling the packet splitting. So just like our `InEndpoint.Read`/`OutEndpoint.Write` accept arbitrary data buffer lengths and expect libusb to split the whole transfer into packets, I'd expect `Control` to do the same.

Can you explain the use case and show the code that you use? Ideally you could also attach the libpcap dump of the USB transfers that you see when you use data length > max packet length.

jennysharps, 2019-06-19:
Thanks for your response -- closing this as I think I was looking at the wrong source & misunderstanding the root of my own issue

closed 2019-06-19 by jennysharps