google/gousb #69: Windows

davidsteed, 2019-06-25:
Got this working on my MAC pretty much straight away.  Have spent hours trying to get it working on Windows.  The instructions are not clear.  Still have no idea what I should be doing to get libusb installed

When I cross compile for windows on my MAC I get lots of compile errors eg. undefined Milliamperes.  Is the windows library different?

zagrodzki, 2019-06-25:
Yes, Windows is different. See https://github.com/google/gousb#notes-for-installation-on-windows. I don't think we'll provide any more support for Windows above that - you need to already be familiar with building POSIX-compatible code on Windows. The instructions for building libusb on Windows are by no means specific to gousb.

closed 2019-06-25 by zagrodzki

mutl3y, 2020-02-25:
thanks to this post I managed to get this working on windows
https://github.com/libusb/libusb/issues/695

For those stumbling across this issue the following worked for me

Windows 10 X64

- Install msys2
- run bash.exe
- install either 32 bit `pacman -S mingw-w64-i686-toolchain` or   64 bit `pacman -Sy mingw-w64-x86_64-toolchain`
- create `/mingw64/lib/pkgconfig/libusb-1.0.pc`
 (example below)
- check version of libusb that installed 
` pacman -Q | grep libusb`
$ mingw-w64-x86_64-libusb 1.0.23-1
- create directory `/mingw64/include/libusb-1.0`
- download correct version of libusb.h and extract to
`/mingw64/include/libusb-1.0/libusb.h`

 
```
#lib-usb-1.0.pc
prefix=/mingw64
exec_prefix=${prefix}
libdir=${exec_prefix}/x86_64-w64-mingw32/lib
includedir=${prefix}/x86_64-w64-mingw32/include

Name: libusb-1.0
Description: C API for USB device access from Linux, Mac OS X, Windows, OpenBSD/NetBSD and Solaris userspace

#update version to match
Version: 1.0.23
Libs: -L${libdir} -llibusb-1.0
Libs.private: 
Cflags: -I${includedir}/libusb-1.0
```