google/gousb #70: Program freezes on gousb.NewContext

xonde, 2019-07-01:
I can `go get` the gousb repo just fine and everything seems to be working, intellisense brings up suggestions and so on.

But when I actually run my program it freezes on "gousb.NewContext()" I tracked it down with breakpoints and found the problem is arising from from the line below

`if err := fromErrNo(C.libusb_init(&ctx)); err != nil {
		return nil, err
	}`

Specifically
`C.libusb_init(&ctx)`

Using the Debugger on VSCode when I "step into" that line above and it goes to a file called `_cgo_gotypes.go` and in the files it says:

`Could not load source '_cgo_gotypes.go'.`

and in the Debug Console it says:

`go list failed on .: can't load package: package .: no Go files in C:\Users\TOJ\AppData\Local\Programs\Microsoft VS Code`

I have absolutely no idea what's going on and would really appreciate if someone could shed some light.

here's my `go env` btw:

```
set GOARCH=amd64
set GOBIN=
set GOCACHE=C:\Users\TOJ\AppData\Local\go-build
set GOEXE=.exe
set GOFLAGS=
set GOHOSTARCH=amd64
set GOHOSTOS=windows
set GOOS=windows
set GOPATH=C:\Work\Important\[REPO-NAME]
set GOPROXY=
set GORACE=
set GOROOT=c:\go
set GOTMPDIR=
set GOTOOLDIR=c:\go\pkg\tool\windows_amd64
set GCCGO=gccgo
set CC=gcc
set CXX=g++
set CGO_ENABLED=1
set GOMOD=
set CGO_CFLAGS=-g -O2
set CGO_CPPFLAGS=
set CGO_CXXFLAGS=-g -O2
set CGO_FFLAGS=-g -O2
set CGO_LDFLAGS=-g -O2
set PKG_CONFIG=pkg-config
set GOGCCFLAGS=-m64 -mthreads -fmessage-length=0 -fdebug-prefix-map=C:\Users\TOJ\AppData\Local\Temp\go-build424489806=/tmp/go-build -gno-record-gcc-switches
```

zagrodzki, 2019-07-01:
I don't think we can help here, this is a problem with your build environment not with gousb.
If it helps, you can take a look at the appveyor configs that we use to build on Windows: https://github.com/google/gousb/blob/master/appveyor.yml

closed 2019-07-01 by zagrodzki

xonde, 2019-07-01:
Thanks for the heads up, would you have any idea what the problem is with my environment? because i'm very stumped right now