google/gousb #71: Zero-length writes are not allowed

QuLogic, 2019-07-29:
In the `transfer` function, a zero-length buffer is [immediately returned](https://github.com/google/gousb/blob/master/endpoint.go#L89) without actually sending anything. However, there seem to be valid use cases for this, so the short-circuit breaks certain things.

I am attempting to port [yubihsm-connector](https://github.com/Yubico/yubihsm-connector/) to gousb from go-libusb, but the lack of empty writes [appears to be problematic](https://github.com/Yubico/yubihsm-connector/issues/3#issuecomment-514199317).

As libusb even includes a flag `LIBUSB_TRANSFER_ADD_ZERO_PACKET` to _automatically_ send empty writes itself (which is not exposed by gousb), I think there's no reason to disallow them from user code and skipping them from the Go layer is incorrect.

zagrodzki, 2019-07-31:
Ok, so if I understand correctly, the only thing needed is the removal of the short-circuit? If an additional zero-length packet is needed after certain transfers (which is what `LIBUSB_TRANSFER_ADD_ZERO_PACKET` would have added automatically), the users will be able to request it manually by just initiating a zero-length write, so we don't need to worry about implementing some support for that transfer flag. Does that sound correct?

@kylelemons I think the code in question was inherited from the previous versions, https://github.com/google/gousb/blame/master/endpoint.go#L89. Can you think of any reason why we shouldn't allow zero-length transfers here?

@QuLogic FYI, in gousb Read/WriteTimeout is instead replaced with a context-aware read/write, see e.g. https://godoc.org/github.com/google/gousb#OutEndpoint.WriteContext

QuLogic, 2019-07-31:
Yes, I believe that is correct. If zero-length writes are allowed, then there's no need to mess with transfer flags (other than convenience, I guess).

I'm aware of the new timeout implementation.

kylelemons, 2019-08-07:
I don't think I had any particular reason; it's possible that I assumed
nonzero somewhere and added it as a guard, but I'm not sure.  Sorry I can't
be more helpful :)

zagrodzki, 2019-08-07:
That works for me, I'll send a PR.

closed 2019-08-12 by zagrodzki