google/gousb #73: cross compile from macOS to linux failed

March-deng, 2019-08-27:
I installed the gcc compiler from `http://crossgcc.rts-software.org/doku.php?id=compiling_for_linux` but I failed to compile the source code to get a exe file.
Any help would be great, thanks.
## code
```golang
package main

import (
	"errors"
	"log"

	"github.com/google/gousb"
)

func main() {
	vid, pid, err := findUsbPrinter()
	if err != nil {
		panic(err)
	}
	log.Println("vid, pid:", vid, pid)
}

func findUsbPrinter() (gousb.ID, gousb.ID, error) {
	ctx := gousb.NewContext()
	defer ctx.Close()

	devs, _ := ctx.OpenDevices(func(device *gousb.DeviceDesc) bool {
		//fmt.Printf("Device %s:%s %s \r\n", device.Vendor, device.Product, usbid.Describe(device))
		for _, cfg := range device.Configs {
			for _, intf := range cfg.Interfaces {
				//fmt.Printf("\tinterface %s\r\n", intf.String())
				for _, setting := range intf.AltSettings {
					if setting.Class == gousb.ClassPrinter {
						return true
					}
				}
			}
		}
		return false
	})
	defer func() {
		for _, d := range devs {
			d.Close()
		}
	}()
	if len(devs) == 0 {
		return gousb.ID(0), gousb.ID(0), errors.New("Can not found printer")
	}
	return devs[0].Desc.Vendor, devs[0].Desc.Product, nil

}

```
## command
```bash
GOOS=linux GOARCH=amd64 CGO_ENABLED=1 CC=/usr/local/gcc-4.8.1-for-linux64/bin/x86_64-pc-linux-gcc go build -v -ldflags="-w -s" -o test-linux main.go
```

## message
```bash
# github.com/google/gousb
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_cancel_transfer':
/tmp/go-build/cgo-gcc-prolog:104: undefined reference to `libusb_cancel_transfer'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_claim_interface':
/tmp/go-build/cgo-gcc-prolog:124: undefined reference to `libusb_claim_interface'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_control_transfer':
/tmp/go-build/cgo-gcc-prolog:163: undefined reference to `libusb_control_transfer'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_detach_kernel_driver':
/tmp/go-build/cgo-gcc-prolog:183: undefined reference to `libusb_detach_kernel_driver'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_get_bus_number':
/tmp/go-build/cgo-gcc-prolog:239: undefined reference to `libusb_get_bus_number'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_get_config_descriptor':
/tmp/go-build/cgo-gcc-prolog:260: undefined reference to `libusb_get_config_descriptor'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_get_configuration':
/tmp/go-build/cgo-gcc-prolog:279: undefined reference to `libusb_get_configuration'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_get_device_address':
/tmp/go-build/cgo-gcc-prolog:297: undefined reference to `libusb_get_device_address'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_get_device_descriptor':
/tmp/go-build/cgo-gcc-prolog:316: undefined reference to `libusb_get_device_descriptor'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_get_device_list':
/tmp/go-build/cgo-gcc-prolog:334: undefined reference to `libusb_get_device_list'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_get_device_speed':
/tmp/go-build/cgo-gcc-prolog:352: undefined reference to `libusb_get_device_speed'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_get_port_number':
/tmp/go-build/cgo-gcc-prolog:370: undefined reference to `libusb_get_port_number'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_get_string_descriptor_ascii':
/tmp/go-build/cgo-gcc-prolog:393: undefined reference to `libusb_get_string_descriptor_ascii'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_handle_events_timeout_completed':
/tmp/go-build/cgo-gcc-prolog:413: undefined reference to `libusb_handle_events_timeout_completed'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_init':
/tmp/go-build/cgo-gcc-prolog:431: undefined reference to `libusb_init'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_open':
/tmp/go-build/cgo-gcc-prolog:450: undefined reference to `libusb_open'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_release_interface':
/tmp/go-build/cgo-gcc-prolog:470: undefined reference to `libusb_release_interface'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_reset_device':
/tmp/go-build/cgo-gcc-prolog:488: undefined reference to `libusb_reset_device'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_set_auto_detach_kernel_driver':
/tmp/go-build/cgo-gcc-prolog:508: undefined reference to `libusb_set_auto_detach_kernel_driver'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_set_configuration':
/tmp/go-build/cgo-gcc-prolog:528: undefined reference to `libusb_set_configuration'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_set_interface_alt_setting':
/tmp/go-build/cgo-gcc-prolog:548: undefined reference to `libusb_set_interface_alt_setting'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_close':
/tmp/go-build/cgo-gcc-prolog:138: undefined reference to `libusb_close'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_exit':
/tmp/go-build/cgo-gcc-prolog:197: undefined reference to `libusb_exit'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_free_config_descriptor':
/tmp/go-build/cgo-gcc-prolog:209: undefined reference to `libusb_free_config_descriptor'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_free_device_list':
/tmp/go-build/cgo-gcc-prolog:223: undefined reference to `libusb_free_device_list'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x004.o: In function `_cgo_3308906fefe0_Cfunc_libusb_unref_device':
/tmp/go-build/cgo-gcc-prolog:576: undefined reference to `libusb_unref_device'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x005.o: In function `gousb_alloc_transfer_and_buffer':
../github.com/google/gousb/transfer.c:81: undefined reference to `libusb_alloc_transfer'
../github.com/google/gousb/transfer.c:87: undefined reference to `libusb_free_transfer'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x005.o: In function `submit':
../github.com/google/gousb/transfer.c:27: undefined reference to `libusb_submit_transfer'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x005.o: In function `gousb_free_transfer_and_buffer':
../github.com/google/gousb/transfer.c:99: undefined reference to `libusb_free_transfer'
/var/folders/6b/5_kf1cw10pz63q53ncx616j40000gn/T/go-build970745123/b003/_x006.o: In function `gousb_set_debug':
../github.com/google/gousb/usb.c:22: undefined reference to `libusb_set_option'
```

zagrodzki, 2019-08-27:
Unfortunately we can't help here, it's not a problem with gousb code.

closed 2019-08-27 by zagrodzki

lhcn, 2022-01-18:
i meet the same issue, and took days to self solved. 

i think might help the others. here's my steps, for your reference:

1. brew install mingw-w64 libusb
2. clone and compile libusb with mingw for windows
3. replace the compiled libs and libusb.h file to  brew install location

😮no more  undefined reference to `libusb_XXXXX'

i don't really know why ,but it's fixed. i guess brew's build is for osx ?

jenish-rudani, 2023-10-28:
> i meet the same issue, and took days to self solved.
>
> i think might help the others. here's my steps, for your reference:
>
> 1. brew install mingw-w64 libusb
> 2. clone and compile libusb with mingw for [...]

This works for compiling binaries for host arm64-apple-darwin23.0.0 (arm64)  and target x86-64 (MS_Windows).
Spent half a day trying to figure this out. This comment helped do this in 5 minutes.

Instructions for cross compilation of gousb:

Steps:

1. brew install minfw-w64 libusb
2. Download Latest Libusb Binaries from [here](https://github.com/libusb/libusb/releases)
3. Copy lib bin include from libusb-x.x.xx-binaries/libusb-MinGW-x64/ to /opt/homebrew/Cellar/libusb/x.x.xx
4. export CC=x86_64-w64-mingw32-gcc #Set Default Compiler
5. export CGO_ENABLED=1 #Enable Cross Compilation
6. export GOOS=windows #Set Windows Env
7. export GOARCH=amd64 #set 64 bit windows architecture (386 for 32bit, you will also need to change CC respectively)
8. go build -v -ldflags="-w -s" -o myTool.exe myTool.go