google/gousb #74: Can you please cut a release or tag ?

alekdu, 2019-10-12:
In order to have others to lock down a reference to this code base, can you please cut a release or tag ? thanks a ton!

zagrodzki, 2020-04-28:
I don't think it's necessary at this point yet - we didn't have (so far) and we don't plan on releasing a backwards-incompatible version in foreseeable future. I thing the rolling model is generally more useful. There are downsides to locking to a specific release, in particular future builds will not receive the bugfixes unless the author spends some effort to see if new releases are available.

If you'd like to use a specific release, I'd recommend vendoring the gousb library in your own project.

@stapelberg and @kylelemons in case they have better ideas.

stapelberg, 2020-04-28:
While not strictly necessary, tagging a version number from time to time is nice in that it gives people an easy-to-compare identifier, and the semantic versioning number itself also conveys the status of the project.

If backwards-incompatible releases aren’t planned, starting with 1.0.0 sounds reasonable to me.

I don’t see a downside in doing this. Do you?

zagrodzki, 2020-04-29:
I personally would worry that users would attach to 1.0.2 at some point and keep using it for the next 5 years... But sure, tags aren't forced on anyone, so why not.

This would be 2.1.0 I supose. 2.0 was the first rewrite of Kyle's library, and I'd say 2.1 happened when streaming endpoints were added. And context operations.

zagrodzki, 2020-04-29:
Done, tagged the current master with v2.1.0.

stapelberg, 2020-04-29:
Wait, if this is v2, then that needs to be reflected in the import path for modules, no? Would be better to tag this v1 (no import path change requirement), I think.

zagrodzki, 2020-04-29:
No? not really? I don't think so. v1 had a different import path (.../kylelemons/gousb vs .../google/gousb). Unless there's some vendoring or other magic that treats "v1.*" and "v2.*" differently.

zagrodzki, 2020-04-29:
I see what you mean with modules and v2 (https://blog.golang.org/v2-go-modules). I don't think it's needed in this case, since this package never had a v1 to begin with. We'd add v3 in a different path though.

stapelberg, 2020-04-29:
What I’m trying to say is that it’s not optional. If you use a v2 tag, you need to use an import path ending in v2. We ran into this in https://github.com/i3/go-i3, but I don’t remember the details.

zagrodzki, 2020-04-29:
I see.

zagrodzki, 2020-04-29:
..and I understand to make that fully usable I'd need to add a `.mod` file saying this is gousb/v2. Which we can do.

zagrodzki, 2020-04-29:
alternatively we could not define a module and users could use https://github.com/golang/go/wiki/Modules#can-a-module-consume-a-v2-package-that-has-not-opted-into-modules-what-does-incompatible-mean. But that also feels wrong.

stapelberg, 2020-04-29:
I think there’s no clean way out of this. If we go with the v2 suffix, users need to update their code.

Personally, my vote is for a v1 tag and no suffix change. Even if the README says that this is 2.0 of kylelemons/gousb, that can be interpreted as a conceptual successor to gousb and doesn’t necessarily mean we can’t start our own versioning.

zagrodzki, 2020-04-29:
But by "users need to update their code" I think you mean only users who refer to gousb in their module dependencies? And since we added the tag only today, there are no existing users that would refer to gousb with a semantic version.
New users that will be using v2.* tag can use a different import path.

What am I missing?

stapelberg, 2020-04-29:
I meant under the assumption that any user of the existing package (`github.com/google/gousb`) today, with or without module mode, should get updates to said package after we tag new releases, without having to manually import package `github.com/google/gousb/v2`.

If users don’t change their code, the Go tool will always stick to the latest non-v2 version, and users might not notice that they’re not using up to date code.

zagrodzki, 2020-04-29:
> the Go tool will always stick to the latest non-v2 version

is that really the case? it sounds very awkward. go get github.com/google/gousb will specifically fetch the last revision before the v2.1.0? I would imagine the default behavior should be "fetch from head", which means all changes are included.

I made a small experiment now and it seems a project that just imports .../gousb will automatically add the requirement:

```
github.com/google/gousb v2.1.0+incompatible
```

and the import path remains unchanged. So perhaps this is all fine?

stapelberg, 2020-04-30:
Maybe. If we take too long to figure this out, we can’t switch anymore. My vote for a v1 version number still stands, but up to you.

zagrodzki, 2020-04-30:
@alekdu, @madpipeline -

zagrodzki, 2020-04-30:
In the meantime I've deleted v2.1.0 and added v1.1.0.

alekdu, 2020-04-30:
Sorry to be later for this thread. I still prefer to have a version or tag for tracking. The idea to keep a software without version and hope everybody always follow latest and greatest version is understandable but, in practice, maybe easy to break compatibility.

Till now, it is still working for me. So V1 version works for me too.

stapelberg, 2020-05-01:
Related: https://github.com/golang/go/issues/38762 (just in case you wanted to read more on the major version suffixes).

zagrodzki, 2020-05-01:
Ok. Since I don't want to spend any more time on this, I'm keeping v1.1.0.

closed 2020-05-01 by zagrodzki