google/gousb #75: Export fakelibusb test code

danielpaulussauce, 2019-10-24:
Can you make the fakelibusb code available for writing unit tests?

zagrodzki, 2020-04-28:
I don't think it's particularly useful outside of gousb, since it fakes a very particular interface internal to gousb (gousb.libusbIntf).

It might be more interesting to offer a fake implementation of gousb.Context - this would be externally visible and could be used as a replacement for gousb.Context injection. It's useful only if you use gousb in your code. Would that be useful?

FWIW, I'm looking into adding a set of integration tests based on Linux virtual USB host driver (vhci-hcd), which I think would offer better fidelity compared to my fake libusbIntf implementation.

danielPaulusMesmer, 2020-11-13:
That sounds good, yeah :-)

zagrodzki, 2020-11-13:
I assume the "sounds good" refers to "offer a fake implementation of gousb.Context". Unfortunately I don't expect to have time to work on that anytime soon. If anyone would like to offer a contribution, I'm happy to help with the design and review.