google/gousb #82: null string in serialNumber issue

jjunghyup, 2020-06-04:
https://github.com/google/gousb/blob/e4c3f66a1571a02d7cae859fa0be08edcb940541/device.go#L246

When there is a null string in the [device.SerialNumber()](https://github.com/google/gousb/blob/e4c3f66a1571a02d7cae859fa0be08edcb940541/device.go#L246) of the [device](https://github.com/google/gousb/blob/master/device.go), I want to get the null string that was stripped.

Serial number example obtained through this library
```
00008020000955DC1132002E\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000
```

Real SerialNumber
```
UniqueDeviceID: 00008020-000955DC1132002E
```

zagrodzki, 2020-06-17:
I think this is working as intended. Per the USB specification (section 9.6.7 in USB 2.0):

> The UNICODE string descriptor (shown in Table 9-16) is not NULL-terminated. The string length is
computed by subtracting two from the value of the first byte of the descriptor.

What you're seeing is a device intentionally setting the string length to 40 bytes (or really, reporting the descriptor `bLength` as 42 bytes, which results in 40 bytes long string). The library has no way of distinguishing intentional use cases where the string descriptor includes a null-character (like the spec says, the string is not null-terminated) from cases where the manufacturer was sloppy.

`libusb` (which `gousb` uses underneath) returns that string unstripped and `gousb` doesn't modify it in any way, and I think this is all as intended. If you know that the null characters are spurious, you should trim the string in your application code.

Can you show, for comparison, the output of `lsusb` with this device?

closed 2020-06-17 by zagrodzki

jjunghyup, 2020-06-18:
Thanks for the good explanation.
If that part is Spec, it might be right not to fix it.

And next is the log of `lsusb` you requested, there is no null value here.(00008020000955DC1132002E)

```
chajunghyeob-xxxxx-SWgisulLab:rtl-ios-capture-demo latte$ lsusb
2020-06-18 11:09:03.677 system_profiler[69358:6421056] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
Bus 020 Device 009: ID 2109:0812 VIA Labs, Inc. USB3.0 Hub 
Bus 020 Device 005: ID 05ac:0274 Apple Inc. Apple Internal Keyboard / Trackpad  Serial: D3H64061QX1GHMFAD6MS
Bus 020 Device 000: ID 05ac:8290 Apple Inc. Bluetooth USB Host Controller 
Bus 020 Device 010: ID 2109:2812 VIA Labs, Inc. USB2.0 Hub 
Bus 020 Device 031: ID 25a7:2410 25a7 Wireless Mouse 
Bus 020 Device 003: ID 05ac:12a8 Apple Inc. iPhone  Serial: 00008020000955DC1132002E
Bus 000 Device 001: ID 1d6b:LPTH Linux Foundation USB 3.0 Bus 
```

zagrodzki, 2020-06-18:
I see the lsusb doesn't have any special handling for trailing spaces/nulls, so I thought it would show the null chars, but now I realized because it uses `printf("...%s...")`, which assumes the string is null-terminated, it will just print the first part of the string up to the null char.

But yes, the spec definitely says the null characters do not indicate the end of the string.