google/gousb #83: ERROR: Unknown revision v2.1.0

HeinzAlberto, 2020-10-13:
Hi there!

 When installing gousb using 
"go get -v github.com/google/gousb" (as written in the Readme on GitHub) I just get the error "go: error loading module requirements"...
(See screenshot.)

Is there a workaround for this problem?
[image]

zagrodzki, 2020-11-05:
Argh. The go proxy (proxy.golang.org) cached the info about v2.1.0, and apparently there's no way to remove it, even though it was present only for one day in April. Sad face.
I've reinstated the v2.1.0.

closed 2020-11-05 by zagrodzki

zagrodzki, 2020-11-09:
FTR, we've released v1.1.1 with a module specification (`go.mod `) and I believe `go get` will prefer that over the v2.1.0+incompatible from now on.