google/gousb #87: A lot of interrupted errors on MacOSX: handle_events: error: libusb: interrupted [code -10]

danielPaulusMesmer, 2020-11-13:
Recently I have seen a lot of these errors: 
`handle_events: error: libusb: interrupted [code -10]` 
on Mac OS X 
with the latest LibUsb
```
libusb: stable 1.0.23 (bottled), HEAD
Library for USB device access
https://libusb.info/
/usr/local/Cellar/libusb/1.0.23 (26 files, 474.2KB) *
```

malc0mn, 2021-12-05:
Just to second this, using `libusb 1.0.24-2` on Manjaro linux and `gousb v1.1.1`

zoracape, 2021-12-12:
Thanks for reporting it. I encountered this issue on Linux and I don't seem to find any problem with USB communication. 
It is probably just a debug log.

Also thanks malc0mn for the suggestion. I just updated libusb.go and disabled this log.

FIS-X, 2022-02-03:
Is there a way to remove this error?

danielpaulus, 2022-04-17:
I am using this `stdlog.SetOutput(new(LogrusWriter))`
and 
```
type LogrusWriter int

const interruptedError = "interrupted [code -10]"

func (LogrusWriter) Write(data []byte) (int, error) {
	logmessage := string(data)
	if strings.Contains(logmessage, interruptedError) {
		log.Tracef("gousb_logs:%s", logmessage)
		return len(data), nil
	}
	log.Infof("gousb_logs:%s", logmessage)
	return len(data), nil
}
```
it is a workaround for Logrus, but you can probably adapt it to whatever you are using.

FIS-X, 2022-04-18:
> I am using this `stdlog.SetOutput(new(LogrusWriter))` where stdlog is the regular golang logger `import stdlog "log"` and
> 
> ```
> type LogrusWriter int
> 
> const interruptedError = "interrupted [code -10]"
> 
> func (LogrusWriter) Write(data []byte) (int, error) {
> 	logmessage := string(data)
> 	if strings.Contains(logmessage, interruptedError) {
> 		log.Tracef("gousb_logs:%s", logmessage)
> 		return len(data), nil
> 	}
> 	log.Infof("gousb_logs:%s", logmessage)
> 	return len(data), nil
> }
> ```
> 
> it is a workaround for Logrus, but you can probably adapt it to whatever you are using.

Thank you!

djadala, 2023-02-01:
Same here in linux:
```github.com/google/gousb/libusb.go:191: handle_events: error: libusb: interrupted [code -10]```

x-abgth, 2023-08-02:
I'm also getting the same error log while erasing the device, and the erasing process is very slow in golang compared to python code. I think the slowness of the erase process is because of the same error.

PS: I converted the python code to golang code manually.

akares, 2023-12-04:
Same here in macOS 14.1.1:
```
2023/12/04 00:58:46 handle_events: error: libusb: interrupted [code -10]
```

commit 2f9ed92c (2024-02-24): suppress 'interrupted [code -10]' log message. resolve #87 (#121)
closed 2024-02-24 by zagrodzki