google/gousb #89: Errors on install; macos

utdrmac, 2021-01-07:
Attempting to follow example in README

```
$ go version
go version go1.13.4 darwin/amd64

$ go get -v github.com/google/gousb/lsusb
/usr/local/go/src/crypto/tls/cipher_suites.go:18:2: code in directory /usr/local/go/src/vendor/golang_org/x/crypto/chacha20poly1305 expects import "golang.org/x/crypto/chacha20poly1305"
/usr/local/go/src/crypto/tls/key_agreement.go:20:2: code in directory /usr/local/go/src/vendor/golang_org/x/crypto/curve25519 expects import "golang.org/x/crypto/curve25519"
/usr/local/go/src/net/http/h2_bundle.go:47:2: code in directory /usr/local/go/src/vendor/golang_org/x/net/idna expects import "golang_org/x/text/internal/export/idna"
/usr/local/go/src/net/http/transport.go:32:2: code in directory /usr/local/go/src/vendor/golang_org/x/net/proxy expects import "golang.org/x/net/proxy"
```

Not sure what to do here

zagrodzki, 2021-01-07:
This looks like a problem with the local installation of Go, not with `gousb` itself.

closed 2021-01-07 by zagrodzki

zagrodzki, 2021-01-07:
For the record, I updated our Travis config to test with Go 1.13 (and also other versions between 1.11 and 1.15), it shows no issues.

utdrmac, 2021-01-07:
Yea, seems I had two versions of Go installed at conflicting paths. I cleaned that up and installed the latest 1.15 and all is well. Thanks.