google/gousb #92: Error when WriteContext is called

txbrown, 2021-01-10:
Hi,

I was using gousb successfully until today. I honestly don't know what happened but now every time I try to run the same code that calls `outEp.WriteContext(ctx, buffer)` I get

```
Assertion failed: (pthread_mutex_lock(mutex) == 0), function usbi_mutex_lock, file ./os/threads_posix.h, line 46.
SIGABRT: abort
PC=0x7fff20353452 m=11 sigcode=0
```

I have code that hasn't been touched in weeks that is now failing but I don't know the reason. I may have upgraded libusb.

zagrodzki, 2021-01-11:
What platform is this on? Can you downgrade libusb to the previous version and compare? The actual version numbers would be useful. Also a small code that reproduces the issue would be helpful.

txbrown, 2021-01-11:
@zagrodzki 
platform: macos 10.15
libusb: 1.0.24 - believe this is the only one available now on homebrew - I had to reinstall my OS 

some code

at this point I already opened the device session and have a handler to it and to the interface.

```go

var abletonVendorID gousb.ID = 0x2982
var pushProductID gousb.ID = 0x1967
var frameHeader = []byte{0xff, 0xcc, 0xaa, 0x88,
	0x00, 0x00, 0x00, 0x00,
	0x00, 0x00, 0x00, 0x00,
	0x00, 0x00, 0x00, 0x00}

var (
	debug = flag.Int("debug", 0, "libusb debug level (0..3)")
)

type AbletonPush2DisplayInterface interface {
	Open() error
	Close() error
}

// Display - Ableton Push 2 Display Interface
type AbletonPush2Display struct {
	pixels []byte
	device *gousb.Device
	ctx    *gousb.Context
	intf   *gousb.Interface
}

func (display *AbletonPush2Display) WritePixels(pixels []uint8) error {
	outEp, _ := display.intf.OutEndpoint(1)
	ctx := context.Background()

	_, err := outEp.WriteContext(ctx, frameHeader)
	_, err := outEp.WriteContext(ctx, pixels)

	return err
}
```

txbrown, 2021-01-11:
@zagrodzki unrelated to this project but the error output is pretty similar to what I have here
https://github.com/moby/moby/issues/37916

txbrown, 2021-01-12:
@zagrodzki I was able to validate that the issue is due to libusb-1.0.24 being installed on my machine.
homebrew is defaulting to that and installing a specific version is really not straightforward.
Could I request support for libusb-1.0.24?

zagrodzki, 2021-01-12:
I'm not sure yet what changes in 1.0.24 triggered this. In the meantime, please fix your code to not ignore the error from the OutEndpoint() as well as from the first WriteContext(). I'd like to see also the rest of the initialization code, i.e. how you obtain the USB context, open the config and claim the interface.

zagrodzki, 2021-01-12:
Also please attach the full output, there's more data in there.

txbrown, 2021-01-12:
@zagrodzki thanks for the feedback. I pretty much went by an example I saw somewhere here on the repo to open device and claim the interface - it's been a while...

But please have a look as I am sure I am doing something really dumb. To be honest when doing this in c++ or wrapping libusb to use in swift I was successful as the api was pretty much what's on libusb docs. On this repo some things feel slightly different and maybe this is because I haven't read enough of the docs - which I am currently going through again slowly.

You can find a repo to reproduce the issue - although without the hardware it would fail running.

https://github.com/txbrown/fyne-push2-example

txbrown, 2021-01-13:
Here is the full console output as well

```
2021/01/13 10:17:21 Device opened
000.001 2982:1967 Unknown 2982:1967
  Protocol: (Defined at Interface level)
Assertion failed: (pthread_mutex_lock(mutex) == 0), function usbi_mutex_lock, file ./os/threads_posix.h, line 46.
SIGABRT: abort
PC=0x7fff202fa462 m=15 sigcode=0

goroutine 0 [idle]:
runtime: unknown pc 0x7fff202fa462
stack: frame={sp:0x70000eb39dc8, fp:0x0} stack=[0x70000eaba350,0x70000eb39f50)
000070000eb39cc8:  0000000000000000  0000000000000064 
000070000eb39cd8:  000000003d2ee000  000070000eb39d50 
000070000eb39ce8:  00007fff2036d575  0000000000000000 
000070000eb39cf8:  000070000eb39d20  0000000004aba766 
000070000eb39d08:  00007fff20280c6f  0000000000000000 
000070000eb39d18:  000070000eb39d68  000070000eb39d60 
000070000eb39d28:  0000000004aba7a6  0000000004aba7b6 
000070000eb39d38:  000000003d2ee000  000000000000002e 
000070000eb39d48:  0000000004aba766  000070000eb39e30 
000070000eb39d58:  00007fff2036b6a7  0000003000000030 
000070000eb39d68:  000070000eb39e40  000070000eb39d80 
000070000eb39d78:  0000000000000000  0000000004ad9a00 
000070000eb39d88:  0000000004adfe00  00007fff889882c0 
000070000eb39d98:  00007fff889aa4e0  00007fff2023470c 
000070000eb39da8:  0000000000000000  000070000eb39df0 
000070000eb39db8:  00007fff20237133  0000000000000000 
000070000eb39dc8: <00007fff20328610  0000000000000003 
000070000eb39dd8:  000000003d2ee028  000070000eb39e10 
000070000eb39de8:  0000000004aba766  000070000eb39e30 
000070000eb39df8:  00007fff2027b720  0000000004aba7a6 
000070000eb39e08:  000000000000002e  00007000fffff9df 
000070000eb39e18:  ffffffff2036b5e4  0000000004aba7a6 
000070000eb39e28:  000000000000002e  000070000eb39e70 
000070000eb39e38:  00007fff2027a9d6  0000000000000016 
000070000eb39e48:  000000c00005dd38  000000000cb0bdd0 
000070000eb39e58:  0000000000000000  000000000c120418 
000070000eb39e68:  0000000000000000  000070000eb39e80 
000070000eb39e78:  0000000004ab918c  000070000eb39ec0 
000070000eb39e88:  0000000004ab2665  00000000042db9d0 
000070000eb39e98:  000000c00005dd38  00000000000000fa 
000070000eb39ea8:  0000000000000000  000000c00005e000 
000070000eb39eb8:  0000000000000000  000070000eb39ef0 
runtime: unknown pc 0x7fff202fa462
stack: frame={sp:0x70000eb39dc8, fp:0x0} stack=[0x70000eaba350,0x70000eb39f50)
000070000eb39cc8:  0000000000000000  0000000000000064 
000070000eb39cd8:  000000003d2ee000  000070000eb39d50 
000070000eb39ce8:  00007fff2036d575  0000000000000000 
000070000eb39cf8:  000070000eb39d20  0000000004aba766 
000070000eb39d08:  00007fff20280c6f  0000000000000000 
000070000eb39d18:  000070000eb39d68  000070000eb39d60 
000070000eb39d28:  0000000004aba7a6  0000000004aba7b6 
000070000eb39d38:  000000003d2ee000  000000000000002e 
000070000eb39d48:  0000000004aba766  000070000eb39e30 
000070000eb39d58:  00007fff2036b6a7  0000003000000030 
000070000eb39d68:  000070000eb39e40  000070000eb39d80 
000070000eb39d78:  0000000000000000  0000000004ad9a00 
000070000eb39d88:  0000000004adfe00  00007fff889882c0 
000070000eb39d98:  00007fff889aa4e0  00007fff2023470c 
000070000eb39da8:  0000000000000000  000070000eb39df0 
000070000eb39db8:  00007fff20237133  0000000000000000 
000070000eb39dc8: <00007fff20328610  0000000000000003 
000070000eb39dd8:  000000003d2ee028  000070000eb39e10 
000070000eb39de8:  0000000004aba766  000070000eb39e30 
000070000eb39df8:  00007fff2027b720  0000000004aba7a6 
000070000eb39e08:  000000000000002e  00007000fffff9df 
000070000eb39e18:  ffffffff2036b5e4  0000000004aba7a6 
000070000eb39e28:  000000000000002e  000070000eb39e70 
[... 108 lines cut ...]
        /Users/ricardoabreu/go/pkg/mod/fyne.io/fyne@v1.4.3/internal/painter/svg_cache.go:107 +0x9c

rax    0x0
rbx    0x70000eb3a000
rcx    0x70000eb39dc8
rdx    0x0
rdi    0x9c13
rsi    0x6
rbp    0x70000eb39df0
rsp    0x70000eb39dc8
r8     0x2e
r9     0xcccccccccccccccd
r10    0x70000eb3a000
r11    0x246
r12    0x9c13
r13    0x3d2ee000
r14    0x6
r15    0x16
rip    0x7fff202fa462
rflags 0x246
cs     0x7
fs     0x0
gs     0x0
exit status 2
```

zagrodzki, 2021-01-14:
I believe this is a bug in your code: you close the context before returning from `AbletonPush2Device.Open` (https://github.com/txbrown/fyne-push2-example/blob/master/display/display.go#L66) - defer is executed as soon as `Open` finishes.

We should probably add better error messages for this case.

txbrown, 2021-01-14:
Omg this is so obvious 🤦🏾‍♂️! I never paid attention to that. Yes that was the issue here. Thanks!

closed 2021-01-15 by zagrodzki