google/gousb #97: Memory Leak

danielpaulus, 2021-04-22:
Hey there, 
I created the minimal repro case for my memory leak.
For some reason memory seems to grow indefinitely when I create a few contexts over the applications lifecycle
in combination with listing devices. 

```
func repro() {
	for {
		ctx := gousb.NewContext()
		devs, _ := ctx.OpenDevices(func(desc *gousb.DeviceDesc) bool {
			return true
		})
		for _, d := range devs {
			d.Close()
		}

		ctx.Close()
	
	}
}
```

Gustavomurta, 2021-04-23:
Hi, 
for what?
I didn't understand your objective. Wrong way, I think. 
Suggested reading: 
https://github.com/google/gousb/blob/master/lsusb/main.go

danielpaulus, 2021-04-23:
Hey @Gustavomurta 
this code is just an example for reproducing the memoryleak faster that I am seeing in my program. 
The code itself should be correct though and although it is not useful for anything other than reproducing the memory leak, it should work just fine. 
If you run this though you would see that the go process keeps using more memory while the golang heap does not increase.

danielpaulus, 2021-04-23:
The same code in Java does not have the issue
```
public static void main(String[] args) {
        while (true) {
            Context context = new Context();
            int errorCode = LibUsb.init(context);
            DeviceList list = new DeviceList();
            LibUsb.getDeviceList(context, list);
            int size = list.getSize();
            for (int i = 0; i < size; i++) {
                DeviceHandle handle = new DeviceHandle();
                LibUsb.open(list.get(i), handle);
                LibUsb.close(handle);
                LibUsb.unrefDevice(list.get(i));
            }
            LibUsb.freeDeviceList(list, false);
            LibUsb.exit(context);
        
        }
    }
```

danielpaulus, 2021-04-23:
I think I found it. 
There seems to be no call to `c.libusb.dereference(dev)` in the leaking case.

closed 2021-11-01 by zagrodzki