google/gousb #102: Latest package not picked up by go get

willmcenaney, 2021-10-21:
Go get is currently prioritising v2.1.0+incompatible over v1.1.1

Do we need to go back up to v2.1.1 to rectify?

zagrodzki, 2021-11-01:
I just tried this on a fresh Ubuntu install:

```
$ /usr/lib/go-1.16/bin/go get -v github.com/google/gousb
go: downloading github.com/google/gousb v1.1.1
github.com/google/gousb
```

seems to work as expected. Can you add steps for reproducing the error and some information about the environment in which you're building? Is it possible that you're using Go 1.13, which didn't ignore `+incompatible` versions? In which case I think you'll have to run `go get github.com/google/gousb@v1` explicitly before you can run `go get -t`. Or upgrade to Go 1.14 or higher.

willmcenaney, 2021-11-04:
Ah yeah that's it - current golang version available on Ubuntu 20.04 repo is 1.13

Thanks for clarifying

mythi, 2022-01-12:
FWIW, `v2.1.0+incompatible` was suddenly picked up by Github's dependabot. Any thoughts what triggers this?

zagrodzki, 2022-01-12:
No. Do you have steps to reproduce? If so, please open a new issue about this.

mythi, 2022-01-12:
It's run by Github automatically so I've not tried to run reproduce it. The logs are:

```
...
updater | INFO <job_254573248> Checking if github.com/google/gousb 1.1.1 needs updating
  proxy | 2022/01/11 20:25:04 [211] GET https://proxy.golang.org:443/github.com/google/gousb/@v/list
  proxy | 2022/01/11 20:25:04 [211] 200 https://proxy.golang.org:443/github.com/google/gousb/@v/list
  proxy | 2022/01/11 20:25:04 [213] GET https://proxy.golang.org:443/github.com/google/gousb/@v/v1.1.1.mod
  proxy | 2022/01/11 20:25:04 [213] 200 https://proxy.golang.org:443/github.com/google/gousb/@v/v1.1.1.mod
  proxy | 2022/01/11 20:25:04 [215] GET https://proxy.golang.org:443/sumdb/sum.golang.org/supported
  proxy | 2022/01/11 20:25:04 [215] 410 https://proxy.golang.org:443/sumdb/sum.golang.org/supported
  proxy | 2022/01/11 20:25:04 [217] GET https://sum.golang.org:443/lookup/github.com/google/gousb@v1.1.1
  proxy | 2022/01/11 20:25:04 [217] 200 https://sum.golang.org:443/lookup/github.com/google/gousb@v1.1.1
  proxy | 2022/01/11 20:25:04 [219] GET https://sum.golang.org:443/tile/8/0/x033/976.p/104
  proxy | 2022/01/11 20:25:04 [219] 200 https://sum.golang.org:443/tile/8/0/x033/976.p/104
  proxy | 2022/01/11 20:25:04 [222] GET https://sum.golang.org:443/tile/8/1/027
  proxy | 2022/01/11 20:25:04 [223] GET https://sum.golang.org:443/tile/8/0/x007/072
  proxy | 2022/01/11 20:25:04 [222] 200 https://sum.golang.org:443/tile/8/1/027
  proxy | 2022/01/11 20:25:04 [223] 200 https://sum.golang.org:443/tile/8/0/x007/072
  proxy | 2022/01/11 20:25:04 [225] GET https://proxy.golang.org:443/github.com/google/gousb/@v/v1.1.1.info
  proxy | 2022/01/11 20:25:04 [225] 200 https://proxy.golang.org:443/github.com/google/gousb/@v/v1.1.1.info
updater | INFO <job_254573248> Latest version is 2.1.0+incompatible
updater | INFO <job_254573248> Requirements to unlock own
updater | INFO <job_254573248> Requirements update strategy 
updater | INFO <job_254573248> Updating github.com/google/gousb from 1.1.1 to 2.1.0+incompatible
...
```

zagrodzki, 2022-03-29:
```
$ go list -m -versions
github.com/google/gousb v1.1.0 v1.1.1 v1.1.2
```

I think this should be solved. Please open a new issue if it still isn't.

closed 2022-03-29 by zagrodzki

mythi, 2022-03-30:
@zagrodzki thanks!

mythi, 2022-04-01:
> ```
> $ go list -m -versions
> github.com/google/gousb v1.1.0 v1.1.1 v1.1.2
> ```
> 
> I think this should be solved. Please open a new issue if it still isn't.

FYI: `dependabot`picked the right version for us. thank you again