google/gousb #107: I can't open hp printer device with gousb

yzzshmily, 2022-03-02:
I have installed libusb, I can find the printer device with gousb, butI can't open the device with OpenDeviceWithVIDPID, what should I do? thanks. 

the  OpenDeviceWithVIDPID return  "Failed to compute node presentation" ,  err is nil .

yzzshmily, 2022-03-03:
[image]

zagrodzki, 2022-03-03:
This seems to be working as intended. You got back two devices. The "Failed to compute" is not an error, it's coming from your IDE.

closed 2022-03-03 by zagrodzki

yzzshmily, 2022-03-03:
[image]

yzzshmily, 2022-03-03:
I use macos, the result is 
[image]

oliverpool, 2024-05-13:
> If error is nil, that means *Device is not nil.

This is not true according to https://pkg.go.dev/github.com/google/gousb#Context.OpenDeviceWithVIDPID

> If none is found, it returns nil and nil error.

(sorry for necrobumping)

zagrodzki, 2024-05-13:
> > If error is nil, that means *Device is not nil.
> 
> This is not true according to https://pkg.go.dev/github.com/google/gousb#Context.OpenDeviceWithVIDPID
> 
> > If none is found, it returns nil and nil error.

indeed, my bad, you're absolutely right. It may return (nil, nil) if no device is found.