google/gousb #111: incorrect device class

zakaria-chahboun, 2022-07-03:
Hi guys! I want to get a list of usb devices, Then catch my *usb printer* by it *class*
So i did this
```go
package main

import (
	"fmt"

	"github.com/google/gousb"
)

func main(){
 
	ctx := gousb.NewContext()
	defer ctx.Close()

	devices, err := ctx.OpenDevices(func(desc *gousb.DeviceDesc) bool{
		fmt.Println(desc.Class)
		return false; // not wanted
	})

	if err != nil {
		panic(err)
	}

	for _, v := range devices {
		fmt.Println(v)
	}
    
}
```
output:
```
hub
hub
wireless
per-interface
```

its gives me `per-interface` instead of "printer" class!

in terminal i get:
```bash
❯ lsusb --verbose | grep "Printer"
  iManufacturer           1 Printer-58
      bInterfaceClass         7 Printer
      bInterfaceSubClass      1 Printer

```

Why? and Thanks!

zagrodzki, 2022-07-03:
This is as expected (presumably). In the Go code you're looking at the class of the device (`Class` field in https://pkg.go.dev/github.com/google/gousb#DeviceDesc). "per-interface" means that different interfaces of this device have different classes. See the description of class `00h` in https://www.usb.org/defined-class-codes#anchor_BaseClass00h

What you're looking at in the `lsusb` output is the interface class of a particular setting of a particular interface on that USB device, which would be reflected in the `Class` field of https://pkg.go.dev/github.com/google/gousb#InterfaceSetting (that's a different type). See the intro to devices, configurations and interfaces in https://pkg.go.dev/github.com/google/gousb#hdr-A_Short_Tutorial.

closed 2022-07-03 by zagrodzki

zakaria-chahboun, 2022-07-04:
So the answer is:
```go
package main

import (
	"fmt"

	"github.com/google/gousb"
)

func main() {
	ctx := gousb.NewContext()
	defer ctx.Close()

	devices, err := ctx.OpenDevices(func(desc *gousb.DeviceDesc) bool {
		for _, config := range desc.Configs {
			for _, inter := range config.Interfaces {
				for _, setting := range inter.AltSettings {
					// Check if the interface class of USB is `Printer`
					if setting.Class == gousb.ClassPrinter {
						return true
					}
				}
			}
		}
		return false
	})

	if err != nil {
		panic(err)
	}

	// Check if printers exists
	if len(devices) == 0 {
		fmt.Println("No Printer Found!")
		return
	}

	fmt.Println("Printers: ")
	for _, device := range devices {
		name, _ := device.Product()
		vid := device.Desc.Vendor
		pid := device.Desc.Product
		port := device.Desc.Port
		fmt.Println(name, vid, pid, port)
	}
}
```

Output:
```bash
❯ go run .
Printers: 
USB Printing Support 0483 070b 2
```

Finally!