google/gousb #112: WinUSB drivers

zakaria-chahboun, 2022-07-04:
Hi!
I want to work with usb devices in Go, On linux i just have to install libsub, But on windows i have to install a lot of building tools, some times it doesn't work properly!
I just want to install `WinUSB` drivers like https://github.com/node-usb/node-usb did! *(it's a corossplatform usb support on javascript node)*

can `gousb` just rely on Zadig drivers without any building steps?

Thanks!

dolmen, 2022-07-04:
If you are using only HID devices, what about package [github.com/karalabe/hid](https://plg.go.dev/github.com/karalabe/hid)? The original author is not responding, but I maintain a [fork](https://github.com/dolmen-go/hid) where I have fixed all the pending issues. The module embeds `libusb` and `hidapi` for cross-platform support without requiring to install drivers. The API is quite limited though.

zakaria-chahboun, 2022-07-05:
@dolmen I actually want to list all USB devices with their *interface class* such as "printer", After getting what i want, It should return to me information like `vid` `pid` `path` etc ...

I don't face any problem with Linux, But on windows it require a lot of building tools, And it have failed!
is you fork works for my case?
Thanks!

zakaria-chahboun, 2022-07-05:
@dolmen 
i have followed these instruction on windows, it's helped me so much!
https://github.com/google/gousb/issues/100

For that i thank god about Linux package managers, Windows is just messy! 🤣