google/gousb #113: libusb: bad access [code -3]

jparrill, 2022-08-02:
I've seen the issue from 2017, but this is still happenning using the example from the Go documentation

```go
// Initialize a new Context.
    ctx := gousb.NewContext()
    defer ctx.Close()

    // Open any device with a given VID/PID using a convenience function.
    // Luxafor (PID 0xf372 | VID 0x04d8)
    dev, err := ctx.OpenDeviceWithVIDPID(0x04d8, 0xf372)
    if err != nil {
        log.Fatalf("Could not open a device: %v", err)
    }
    defer dev.Close()

    cfg, err := dev.Config(1)
    if err != nil {
        log.Fatalln("Cannot instanciate Config from Dev")
    }

    fmt.Println(cfg.Interface(0, 0))

    // Claim the default interface using a convenience function.
    // The default interface is always #0 alt #0 in the currently active
    // config.
    intf, done, err := dev.DefaultInterface()
    if err != nil {
        log.Fatalf("%s.DefaultInterface(): %v", dev, err)
    }
    defer done()
```

I'm trying to use it with Luxafor device, which are those VID,PID.

And this is the output:

```
<nil> failed to claim interface 0 on vid=04d8,pid=f372,bus=4,addr=8,config=1: libusb: bad access [code -3]
2022/08/02 18:23:58 vid=04d8,pid=f372,bus=4,addr=8.DefaultInterface(): failed to select interface #0 alternate setting 0 of config 1 of device vid=04d8,pid=f372,bus=4,addr=8: failed to claim interface 0 on vid=04d8,pid=f372,bus=4,addr=8,config=1: libusb: bad access [code -3]
exit status 1
```

It's failling on the default interface assignation: `intf, done, err := dev.DefaultInterface()`

I've tried other number as an num of inerface but the output shows that the `0, 0` values are the right ones.

```
<nil> descriptor of interface (3, 0) in vid=04d8,pid=f372,bus=4,addr=8,config=1: interface 3 not found, available interface numbers: [0]
```

```
<nil> descriptor of interface (0, 3) in vid=04d8,pid=f372,bus=4,addr=8,config=1: alternate setting 3 not found for Interface 0 (1 alternate settings), available alt settings: [0]
```

Any ideas?

jparrill, 2022-08-31:
Hello?

chrBrd, 2022-09-05:
> I've seen the issue from 2017

Have you tried [the solution](https://github.com/google/gousb/issues/21#issuecomment-333363944) from that thread? This is almost definitely a permissions error at your end, not a library problem.

As also mentioned in that issue, you can add `ctx.Debug(4)` to your code before the call to `OpenDeviceWithVIDPID()` to get libus's debug output, which is likely to contain something like:

```
libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/004/008, errno=13
libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
libusb: debug [libusb_open] open 4.8 returns -3
```

jparrill, 2022-09-05:
Adding the Debug statement, I've seen this:

```
[ 0.014493] [000663a0] libusb: debug [libusb_get_configuration] active config 1
vid=04d8,pid=f372,bus=4,addr=3,config=1
[ 0.014647] [000663a0] libusb: debug [libusb_claim_interface] interface 0
[ 0.015079] [000663a0] libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
<nil> failed to claim interface 0 on vid=04d8,pid=f372,bus=4,addr=3,config=1: libusb: bad access [code -3]
[ 0.015166] [000663a0] libusb: debug [libusb_get_configuration]
[ 0.015169] [000663a0] libusb: debug [libusb_get_configuration] active config 1
[ 0.015173] [000663a0] libusb: debug [libusb_get_configuration]
[ 0.015175] [000663a0] libusb: debug [libusb_get_configuration] active config 1
[ 0.015178] [000663a0] libusb: debug [libusb_claim_interface] interface 0
[ 0.015419] [000663a0] libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
2022/09/05 22:26:07 vid=04d8,pid=f372,bus=4,addr=3.DefaultInterface(): failed to select interface #0 alternate setting 0 of config 1 of device vid=04d8,pid=f372,bus=4,addr=3: failed to claim interface 0 on vid=04d8,pid=f372,bus=4,addr=3,config=1: libusb: bad access [code -3]
exit status 1
```

jparrill, 2022-09-06:
@chrBrd any ideas about the trace I've copied? I mean, looks like there are an info message just before the Error saying something about exclusive access to the device but I didn't modify anything in the permissions side. Even I've executed the Go code with sudo and is getting me the same error...

chrBrd, 2022-09-06:
Apparently it's caused by Apple locking down device access in macOS 12. There's a long thread on an [open issue](https://github.com/libusb/libusb/issues/972) in the libusub repo. There's a [workaround](https://github.com/libusb/libusb/wiki/FAQ#How_can_I_run_libusb_applications_under_Mac_OS_X_if_there_is_already_a_kernel_extension_installed_for_the_device_and_claim_exclusive_access) suggested in libusb's FAQ, and a more straight-to-the-point [digest of it](https://stackoverflow.com/a/29934989/3808078) on Stack Overflow.

chrBrd, 2022-09-08:
@jparrill As this issue isn't anything to do with gousb, could you please do the maintainers a wee favour and close it?

closed 2022-09-08 by jparrill

Dynom, 2023-04-12:
Hi,

I've a similar issue on a RaspberryPI and I'm seeing different behaviour between usage and while it might not be a problem with gousb, I do hope anyone can point me in the right direction.

Trying to open a USB device, I see the infamous `libusb: bad access [code -3]`. Running the same application elevated and it all works as expected: I'm able to "open" the device at least.
However what I don't understand is why I'm unable to open the USB device.

I run the application as a user, member of the group `gpio` and I've added liberal permissions to the device:
```
pi@raspberrypi:/dev/input $ l
crw-rw-rw- 1 root gpio  13, 66 Apr 12 15:47 event2
```

Is the library not leaning on udev? And/or how can I give access to a device so that libusb is able to access it without elevated privileges?

Thanks in advance.

Gustavomurta, 2023-04-13:
Hi Dynom .
Follow my modest suggestion for Raspberry Pi installation: 

**Updated Gousb instructions for Raspberry Pi : (2021)** 
https://github.com/google/gousb/issues/45#issuecomment-945317497