google/gousb #115: Invalid param [code -2] for iOS 16 devices

Karazum, 2023-02-14:
Hello,

I'm running an application that mirrors iphone screen via usb cable to my linux desktop.
The application in question is a modified [quick time video hack](https://github.com/Karazum/quicktime_video_hack)
Essentially, when a iphone is connected to a linux machine, it determines which usb configuration is for audio-video mirroring and when you run the application it will attempt to switch to that usb configuration but it fails with the following error:
{"error_message":"failed connecting to usb","original_error":"Can't detach kernel driver of the device vid=05ac,pid=12a8,bus=5,addr=21 and interface 3: **libusb: invalid param [code -2**]"}

Additional useful information:
- autodetach is enabled
- linux version: ubuntu 22.04
- usbmuxd version: 1.1.2
- steps to reproduce:
    - git clone https://github.com/Karazum/quicktime_video_hack.git
    - go build -o qvh
    - plug in iphone with iOS 16 or more (iOS below 16 works)
    - qvh devices (to check if the device is available)
    - qvh gstreamer

GeoffAbbott, 2023-03-08:
Did you ever make any progress in this? Running into same issue with same qvh library. Also running into libusb: device or resource busy [code -6] when not attempting to auto detach kernal.

Karazum, 2023-03-08:
Yes. It seems that the issue was in the usbmuxd binary.
Downloading the latest version from git and manually building the binary resolves the issue.

GeoffAbbott, 2023-03-09:
That fixed the issue, Thanks!! Stuck on the "waiting for Ping..." Were you able to start receiving packets?

GeoffAbbott, 2023-03-09:
Wow was able to get that to work as well. Thanks again!