google/gousb #119: How to use in a Docker container

half2me, 2023-08-30:
Is there any example for this being used inside a docker image? I have an app which uses this to speak to a USB device, but as soon as I run my application in a container it no longer finds the device.
I've tried using the `--privileged` mode etc, but I can't figure out how to make it work. Do I need to use the `--device` option? Or is there some path I need to mount in to get this to work?

frelon, 2024-01-11:
Hi @half2me, I just stumbled upon this issue and I have been using the following `docker run --device=/dev/bus/usb/<bus>/<device>` to share the device without the `--privileged` flag.

Hope it helps!

dhkimxx, 2024-11-10:
Or Try This: `docker run -v /dev/bus:/dev/bus:ro -v /dev/serial:/dev/serial:ro ...`