google/gousb #120: CDC device

JetSetIlly, 2023-10-14:
Hello,

I have a STM32 microcontroller (bluepill) that I have successfully set up with a small program to emit data on the USB port. When connected with the cu program (connecting to the /dev/ttyACM0 device) I receive the stream of data. This proves that the bluepill is working as I expect.

I am now trying to use gousb to retrieve the data. I am using the example programs from the package's documentation as a basis for a test program. The logs indicate that I am finding the correct device and successfully opening the In endpoint.

The problem seems to be that the in.Read() function never returns. I see this debug message on the call to Read:

> [ 0.192353] [00009985] libusb: debug [libusb_alloc_transfer] transfer 0x2232898
> [ 0.192379] [00009985] libusb: debug [libusb_submit_transfer] transfer 0x2232898
> [ 0.192392] [00009985] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 128
> [ 0.206358] [00009987] libusb: debug [usbi_wait_for_events] poll() returned 0

and then repeated messages thereafter:

> [ 0.206395] [00009987] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
> [ 0.206403] [00009987] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 100ms
> [ 0.306535] [00009987] libusb: debug [usbi_wait_for_events] poll() returned 0

I'm not sure how to proceed from here. Permissions to the usb device are correct and the device itself is functioning as I expect. Is there a configuration step that I have misunderstood?

I include the complete test program as a gist for reference:  https://gist.github.com/JetSetIlly/52ae2f639f7d524a1fa58dac91b950ff

JetSetIlly, 2023-10-15:
I've also tried the rawread program (from the gousb repository) and ran it with the appropriate interface/endpoint arguments. I receive the same log messages.

zagrodzki, 2023-10-16:
sorry, I don't think we can help with such development issues. That being said, if you can see that a C/libusb program works as expected while gousb doesn't, then it's possible the problem lies within gousb and I would take a look.

zagrodzki, 2023-10-16:
my personal guess would be that the device needs some kind of setup before it starts transmitting - perhaps the right configuration needs to be selected on the device, or some control command needs to be sent before endpoint data is produced.

closed 2023-10-16 by zagrodzki

JetSetIlly, 2023-10-16:
Understood. I now think it's probably a USB hardware issue on the micro controller.

zagrodzki, 2023-10-16:
using Wireshark to snoop on USB packets in a case where it does and doesn't work may be enlightening.