google/gousb #128: Error claiming interface

michealroberts, 2024-06-08:
I'm perhaps looking more for advice on what is going wrong, but I am seeing this error:

```
Error claiming interface: descriptor of interface (1, 1) in vid=1278,pid=0525,bus=3,addr=36,config=1: alternate setting 1 not found for Interface 1 (1 alternate settings), available alt settings: [0]
```

This is my Connect functionality:

```
func (d *Driver) Connect(vendorId gousb.ID, productId gousb.ID, configuration DeviceConfiguration) error {
	ctx := gousb.NewContext()

	ctx.Debug(4)

	defer ctx.Close()

	device, err := ctx.OpenDeviceWithVIDPID(vendorId, productId)
	if err != nil {
		return err
	}
	if device == nil {
		return fmt.Errorf("device not found")
	}
	defer device.Close()

	if len(device.Desc.Configs) < 1 {
		log.Fatalf("Device does not have any configurations")
	}

	d.device = &Device{ctx: ctx, handle: device}

	cfg, err := device.Config(configuration.Number)

	if err != nil {
		log.Printf("Error getting device config: %s", err.Error())
		return err
	}

	defer cfg.Close()

	for _, intf := range cfg.Desc.Interfaces {
		if len(intf.AltSettings) == 0 {
			log.Printf("No alternate settings available for interface %d", intf.Number)
			return fmt.Errorf("no alternate settings available for interface %d", intf.Number)
		}

		alt := intf.AltSettings[0]

		log.Printf("Processing interface %d alternate setting %d", intf.Number, alt.Number)

		iface, err := cfg.Interface(intf.Number, alt.Number)

		if err != nil {
			log.Printf("Error claiming interface: %s", err.Error())
			return err
		}

		defer iface.Close()

		for _, ep := range alt.Endpoints {
			if ep.Direction == gousb.EndpointDirectionOut {
				outEp, err := iface.OutEndpoint(ep.Number)
				if err != nil {
					log.Printf("Error getting device out endpoint: %s", err.Error())
					return err
				}
				d.device.outEndpoint = outEp
				fmt.Printf("Found OUT endpoint: %d\n", ep.Number)
			} else if ep.Direction == gousb.EndpointDirectionIn {
				inEp, err := iface.InEndpoint(ep.Number)
[... 104 lines cut ...]
[ 0.003226] [00d6bdce] libusb: debug [libusb_unref_device] destroy device 3.31
[ 0.003227] [00d6bdce] libusb: debug [libusb_unref_device] destroy device 3.30
[ 0.003229] [00d6bdce] libusb: debug [libusb_unref_device] destroy device 3.29
[ 0.003230] [00d6bdce] libusb: debug [libusb_unref_device] destroy device 3.9
[ 0.003231] [00d6bdce] libusb: debug [libusb_unref_device] destroy device 3.5
[ 0.003232] [00d6bdce] libusb: debug [libusb_unref_device] destroy device 3.4
[ 0.003233] [00d6bdce] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.003888] [00d6bdce] libusb: debug [usbi_remove_event_source] remove fd 3
2024/06/08 15:36:21 failed to connect to device: descriptor of interface (1, 1) in vid=1278,pid=0525,bus=3,addr=36,config=1: alternate setting 1 not found for Interface 1 (1 alternate settings), available alt settings: [0]
exit status 1
```

I think the device can be connected to, it's just that the alternative interface numbers are not matching up ...

When I manually change to this:

```
iface, err := cfg.Interface(1, 0)
```

I get an index out of range panic ... 

I am stumped. Is this a firmware issue?

Let me know what other information is needed, if any ...

michealroberts, 2024-07-07:
@krasin Is there a way to call `libusb_detach_kernel_driver()` to ensure that we can claim the target interface?

michealroberts, 2024-07-07:
cc @kylelemons Not sure who to ping ...

krasin, 2024-07-07:
> @krasin Is there a way to call `libusb_detach_kernel_driver()` to ensure that we can claim the target interface?

@michealroberts it's being 11 years since I touched this codebase. Sorry, but my memory returns zero context about it...

michealroberts, 2024-07-09:
@krasin No worries, trying to find the core maintainers for this ... if any are left? :(

zagrodzki, 2024-07-16:
Before I review your PR #129, I'd like to see some data to confirm where the issue comes from. Post a result of `lsusb -v` for the device in question.

closed 2024-09-13 by zagrodzki

zagrodzki, 2024-09-13:
For future reference, the discussion happened mostly in the comment threads of #129 .

The `lsusb` output of the problematic device:

```
002.001 1278:0525 Unknown (Starlight Xpress)
  Protocol: (Defined at Interface level)
  Configuration 1:
    --------------
    Interface 1 alternate setting 0 (available endpoints: [0x01(1,OUT) 0x82(2,IN)])
      Vendor Specific Class
      ep #2 IN (address 0x82) bulk [512 bytes]
      ep #1 OUT (address 0x01) bulk [512 bytes]
    --------------
```

and the problem was that the interface number here is not 0 (as expected by the USB for the first interface).

There was an additional problem with the code mentioned in https://github.com/google/gousb/issues/128#issue-2341686861 - that code used `intf.Number` as the interface number and `alt.Number` as the alternate setting number. This is a mistake, since `alt.Number` is always the same as the interface number - the alternate number in the example code should be `alt.Alternate`. This is why the code tried to claim `Interface(1,1)` rather than `Interface(1,0)` and this is why this failed - there was no alternate setting 1. This was as intended. Like reported, calling `Interface(1,0)` directly resulted in out of bounds access, as `Config.Interfaces` had only a single element (a single interface in the device above), but its `Number` was 1, not 0.